stereolabs/zed-python-api #125: I install following the instrcution on github, but it always has the error "ModuleNotFoundError: No module named 'pyzed.sl'"

hzm2016, 2020-01-31:
ZED SDK Version: OK
compilation flags: ['-std=c++14', '-Wno-reorder', '-Wno-deprecated-declarations', '-Wno-cpp', '-O3']
include dirs: ['/home/zhimin/.local/lib/python3.6/site-packages/numpy/core/include', '/usr/local/zed/include', '/usr/local/cuda/include']
library dirs: ['/home/zhimin/.local/lib/python3.6/site-packages/numpy/core/include', '/usr/local/zed/lib', '/usr/local/cuda/lib64']
libraries: ['sl_zed', 'usb-1.0']
Building module: ('pyzed.sl', ['pyzed/sl.pyx'])
running build
running build_py
creating build
creating build/lib.linux-x86_64-3.6
creating build/lib.linux-x86_64-3.6/pyzed
copying pyzed/__init__.py -> build/lib.linux-x86_64-3.6/pyzed
running build_ext
building 'pyzed.sl' extension
creating build/temp.linux-x86_64-3.6
creating build/temp.linux-x86_64-3.6/pyzed
gcc -pthread -B /home/zhimin/anaconda3/envs/tensorflow/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I./pyzed -I/home/zhimin/.local/lib/python3.6/site-packages/numpy/core/include -I/usr/local/zed/include -I/usr/local/cuda/include -I/home/zhimin/anaconda3/envs/tensorflow/include/python3.6m -c pyzed/sl.cpp -o build/temp.linux-x86_64-3.6/pyzed/sl.o -std=c++14 -Wno-reorder -Wno-deprecated-declarations -Wno-cpp -O3
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
g++ -pthread -shared -B /home/zhimin/anaconda3/envs/tensorflow/compiler_compat -L/home/zhimin/anaconda3/envs/tensorflow/lib -Wl,-rpath=/home/zhimin/anaconda3/envs/tensorflow/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.6/pyzed/sl.o -L/home/zhimin/.local/lib/python3.6/site-packages/numpy/core/include -L/usr/local/zed/lib -L/usr/local/cuda/lib64 -Wl,-R/home/zhimin/.local/lib/python3.6/site-packages/numpy/core/include -Wl,-R/usr/local/zed/lib -Wl,-R/usr/local/cuda/lib64 -lsl_zed -lusb-1.0 -o build/lib.linux-x86_64-3.6/pyzed/sl.cpython-36m-x86_64-linux-gnu.so

Poutsh, 2020-02-04:
Hi @hzm2016 

This looks like the logs during the build of the pyzed module
Have you done those two commands :
```
python3 setup.py build
```
And then :
```
python3 setup.py install
```

hzm2016, 2020-02-07:
I have solved it, the file is not 'sl.so' , I need to revise the name of the file in '/site-packages'.

chikko80, 2020-05-06:
> I have solved it, the file is not 'sl.so' , I need to revise the name of the file in '/site-packages'.

Hello, could you please explain what exactly you have done? i dont have a .so file - i am stuck for days now on this problem and couldnt solve it.

noelquah, 2020-07-03:
I had a similar problem. I noticed that i was installing pyzed on my non-virtual environment while my IDE was defaulted to a python version from a virtual environment, which did not have pyzed installed during the process. Solved the issue by rerouting to the correct python file path.

closed 2022-04-27 by github-actions[bot]