stereolabs/zed-python-api #126: module 'pyzed.sl' has no attribute 'OBJECT_TRACK_STATE'

TaynaCT, 2020-02-04:
So I'm building an app in python for the ZED 2 using  Objects detection module. O the main thread I try to see the OBJECT_TRACK_STATE. But once in a while, when the camera detects a lot of things, the program throws this error message:

``` Python
Exception in thread Thread-3:
Traceback (most recent call last):
  File "C:\Users\unloo\AppData\Local\Programs\Python\Python37\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Users\unloo\AppData\Local\Programs\Python\Python37\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\unloo\PycharmProjects\test\manager\zed\zed_manager.py", line 137, in start_async
    if i < self.player_num or object_state != sl.OBJECT_TRACK_STATE.OK:
AttributeError: module 'pyzed.sl' has no attribute 'OBJECT_TRACK_STATE'
```

What is going on? 
These are the conditions I have on my code:

``` Python
                    if object_id is 4294967295:
                        continue

                    if object_state is None:
                        continue

                    if self.data is None:
                        if i < self.player_num or object_state != sl.OBJECT_TRACK_STATE.OK:
                            self.data = bytes(('{"id": ' + str(object_id) + ', "tracking_state": "' + str(object_state) +
                                               '", "position": [' + str(object_position[0]) + ',' +
                                               str(object_position[1]) + ',' + str(object_position[2]) + ']}'), 'UTF-8')

``` 

#zed2  #pyzed.sl

Poutsh, 2020-02-04:
Hi @TaynaCT ,

In SDK 3.0, there is no ```OBJECT_TRACK_STATE``` but there is [OBJECT_TRACKING_STATE](https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1OBJECT__TRACKING__STATE.html) instead.

TaynaCT, 2020-02-04:
Hi @Poutsh ,

apparently this was the problem. Thank you. 
But just FYI, on this page( [https://www.stereolabs.com/docs/object-detection/using-object-detection/](url) ) on the ACCESSING OBJECT INFORMATION section, they put wrong info :( The example they give is:

```Python
if object_tracking_state == sl.OBJECT_TRACK_STATE.OK :
    print("Object {0} is tracked\n".format(object_id))
```

it shoud be `sl.OBJECT_TRACKING_STATE.OK` instead.

Than you for your help. :)

closed 2020-02-04 by TaynaCT