stereolabs/zed-python-api #129: Error extracting (x, y, z) values from pointcloud data

avnishks, 2020-02-06:
Hi,

I have a list of specific pixels and I am extracting their corresponding _world-coordinates (x, y, z)_ from the ZED video, by using:
`err, point_cloud_value = point_cloud.get_value(x_pixel, y_pixel)`
This script runs fine on most of my SVO files but in  some cases throws up the following error:
> File "pyzed/sl.pyx", line 2077, in pyzed.sl.Mat.get_value
> OverflowError: can't convert negative value to size_t

Anyone else encounter this before? Any pointers on what might be going wrong would be helpful!
Thanks!

saratrajput, 2020-03-07:
I believe the point cloud value might not be available for that particular pixel, so it returns a -1 value. 

You can have a look at this [function](https://github.com/stereolabs/zed-yolo/blob/13ea7f1dafb1fe734a1b88b53c3335fb594b6193/zed_python_sample/darknet_zed.py#L264-L300) in [zed-yolo](https://github.com/stereolabs/zed-yolo) in how to extract depth for corresponding pixel values.

Basically you take a range of depth values for the corresponding 2D coordinates and take their median. And any unavailable depth values will be ignored.

You can also try to use the *FILL* Depth Mode, so there won't be many missing depth values.

avnishks, 2020-03-12:
Thanks for the suggestion @saratrajput . I am infact using the _FILL_ depth mode while reading my SVO file:
`runtime_parameters.sensing_mode = sl.SENSING_MODE.SENSING_MODE_FILL`

Is it possible that the _FILL_ mode has some missing depth values as well?

saratrajput, 2020-03-16:
It could if the depth is out of range. You can configure your ```depth_minimum_distance``` and ```depth_maximum_distance```. Also there could be some issues for dark objects or reflections. You could try to visualize it using ```Zed_Depth_Viewer```.

closed 2022-04-27 by github-actions[bot]