stereolabs/zed-python-api #132: Texture Applied doesn't work in SDK 3.01, 3.02 (Python API)

prelee, 2020-02-12:
Good day, I am using the Python API to create a Spatial Map using a Zed Mini.

The spatial map that is generated appears geometrically close to correct, however when applying textures I have the following problems:
1) Using SDK 3.01, the applied textures do not apply properly, it seems the overall png file is being applied, not the individual tiles to the appropriate face.
2) When updating to SDK 3.02, the apply_texture function fails and no texture is applied.

I've gone back to confirm if this problem crops up on previous SVO files that I have created spatial maps from successfully and it does.

Using ZedFu on my computer applies textures appropriately to the model.

Could you please let me know if I'm doing something wrong?

prelee, 2020-02-13:
Picture of the texture applied to a portion of the model demonstrating the issue:

[image]

Settings on this run: 
sl.DEPTH_MODE.PERFORMANCE
sl.SENSING_MODE.STANDARD
sl.UNIT.METER
sl.MESH_FILTER.MEDIUM
sl.MESH_TEXTURE_FORMAT.RGB

adujardin, 2020-02-13:
Could you share a sample and if possible an SVO file that triggers this issue? You can share it privately by email at `support@stereolabs.com` if you prefer. We weren't able to reproduce it with the same parameters and version.

Could you also share the configuration of your PC please (jetson/desktop, Jetpack version, OS, CPU, RAM, GPU...) ? You can run the ZED Diagnostic and upload the json result file.

prelee, 2020-02-17:
Thanks for your response adujardin, I've been experimenting with it over the last few days, it seems to occur when I have a subfolder that contains the same obj/mtl file names, and only seems to be an issue in 3D viewer.  Based on this I don't believe it is a Zed / Stereolabs issue, so I'll close this and bring it back up if there seems to be some other issue.  

Thanks for your work on the Zed device and code it is quite interesting for our application if we could just get the mesh to be a bit more consistent.

closed 2020-02-17 by prelee