stereolabs/zed-python-api #134: Recording SVO causes libpng error and eventual segfault

dustymugs, 2020-02-17:
Using `stereolabs/zed:3.0-gl-devel-cuda10.2-ubuntu18.04` docker image (image id 22cca9a03721), I did the following:

Launch docker container
```
docker run -it --gpus all --privileged -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -e NVIDIA_DRIVER_CAPABILITIES="video,compute,utility" stereolabs/zed:3.0-gl-devel-cuda10.2-ubuntu18.04
```

In the docker container
```
apt-get update
apt-get install -y libqt5opengl5 libq5svg5

/usr/local/zed/tools/ZED_Explorer # to make sure the camera works
```

Install python3
```
apt-get install -y python3 python3-pip git
pip install cython numpy ipython ipdb
cd /opt
git clone https://github.com/stereolabs/zed-python-api
cd zed-python-api
python3 setup.py build
python3 setup.py install
cd /usr/local/lib/python3.6/dist-packages/pyzed
ln -s sl.cpython-36m-x86_64-linux-gnu.so sl.so
python -c "import pyzed.sl as sl"  # confirm that zed python api works
```

Run the following test script

```
import pyzed.sl as sl
import os
from signal import signal, SIGINT
import sys

output_path = '/tmp/foo.svo'

def handler(signal_received, frame):
    zed.disable_recording()
    zed.close()
    sys.exit(0)

signal(SIGINT, handler)

init_params = sl.InitParameters()
init_params.camera_resolution = sl.RESOLUTION.HD720
init_params.camera_fps = 15

zed = sl.Camera()
err = zed.open(init_params)
if (err != sl.ERROR_CODE.SUCCESS):
    print(repr(err))
    exit(-1)

record_params = sl.RecordingParameters(
    video_filename=output_path,
    compression_mode=sl.SVO_COMPRESSION_MODE.LOSSLESS
)
err = zed.enable_recording(record_params)
if (err != sl.ERROR_CODE.SUCCESS):
    print(repr(err))
    exit(-1)

i = 0
while True:
    if zed.grab() == sl.ERROR_CODE.SUCCESS:
        if i % 100 == 0:
            dropped = zed.get_frame_dropped_count()
            print(
                    '{} / {} ({:.2f} %)'.format(
                    dropped,
                    i,
                    (
                        (dropped / i) * 100.
                        if i > 0
                        else 0.
                    )
                ),
                flush=True
            )
        i += 1
```

The test scripts results in a segfault after a few seconds.

Putting a breakpoint in the script before the `while True` statement and stepping through, I get a flood of the following:

```
libpng warning: Image width exceeds user limit in IHDR
libpng warning: Image height is zero in IHDR
libpng error: Invalid IHDR data
```

As of right now, I can't use the python api to record video...

adujardin, 2020-02-18:
I tried to reproduce the issue. 
Since I'm running CUDA 10.0 with the associated driver I was only able to run the `stereolabs/zed:3.0-gl-devel-cuda10.0-ubuntu18.04` image, but I tried different resolutions, cameras, and parameters and it worked fine for each. 

I was thinking about a libpng version conflict but this issue usually prints warning explicitly saying it.
The warning you're getting usually indicates an issue with the camera opening/setup. But the SDK checks these things during the open() and shouldn't return SUCCESS if something isn't correct.

Some things you can try to identify the source of this issue :
- If possible not using docker, installing natively the SDK and python wrapper
- Testing other docker images (like Ubuntu 16 and maybe also other versions of CUDA)
- Testing to record with the same parameters but from the [C++ sample](https://github.com/stereolabs/zed-examples/tree/master/svo%20recording/recording)
- Testing other SVO compression mode (H264/265)

Note: The `NVIDIA_DRIVER_CAPABILITIES` should also have 'graphics' to be able to run OpenGL programs. Just in case for the testing and to be able to use the hardware encoder, `-e NVIDIA_DRIVER_CAPABILITIES=all` can be used.

dustymugs, 2020-02-18:
I reverted back to zed sdk 2.8 (docker image 40e7577bf374) and corresponding zed-python-api (branch zedsdk_2.X) and all the problems went away. More disturbingly was that the entire reason for this investigation (a serious frame rate drop only when recording) also cleared upon when reverting back to 2.8.

I'll try a few other docker images to see if the behavior changes

adujardin, 2020-02-18:
The lossless compression is heavy on the CPU since every image is compressed into png. That's why the SVO files are so big and require fast storage mediums, and also why a lot of frame drops can occur.

If possible you should really consider using the H264/265 compression which considerably reduces the SVO size (and storage throughput requirements) and CPU usage since the compression is using hardware acceleration.

dustymugs, 2020-02-18:
Isn't it still a performance regression either way given similar settings on two different versions of the sdk resulting in two very different behaviors?

And H264/265 would be great if my hardware supported it

adujardin, 2020-02-18:
A performance regression is always possible, could you give more details to be able to reproduce it, please?

If your hardware supports the ZED SDK then it should also support at least H264 encoding (except in some very rare cases like the GT1030 or the Quadro P500, https://developer.nvidia.com/video-encode-decode-gpu-support-matrix). 
Using docker it could be disabled depending on the `NVIDIA_DRIVER_CAPABILITIES` given. 

What is your configuration?

dustymugs, 2020-02-18:
I'm running on this laptop [UX303UB](https://www.asus.com/us/Laptops/ASUS-ZenBook-UX303UB/). Just tried using AVCHD and HEVC on the docker container and both result in `SVO UNSUPPORTED HARDWARE FOR COMPRESSION`.

nvidia-smi works in the docker container resulting in...

```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.59       Driver Version: 440.59       CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce 940M        Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   38C    P0    N/A /  N/A |      0MiB /  2004MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|  No running processes found                                                 |
+-----------------------------------------------------------------------------+
```

Shame that AVCHD and HEVC via Intel QuickSync isn't supported as I would absolutely use that.

Part of my problem is I am using the zed camera out in the field for agricultural use cases carrying a backpack and won't be back in range for power recharge or internet for many hours. Even using the GPU for compressing is cause for concern due to the power draw.

adujardin, 2020-02-18:
From the Nvidia documentation, the 940M should support H264/AVCHD encoding (the 940MX variant might not). 
Did you try with `-e NVIDIA_DRIVER_CAPABILITIES=all` to run the docker image?

The hardware encoding use dedicated chips that are much more power-efficient than the GPU itself. You should see improved power usage when using `NVENC` compared to CPU encoding. Our test showed improved battery life on laptops when recording SVO in H264/5.

dustymugs, 2020-02-18:
Tried with `-e NVIDIA_DRIVER_CAPABILITIES=all` and no success. Definitely not supported as `lspci` gives:

```
...
01:00.0 3D controller: NVIDIA Corporation GM108M [GeForce 940M] (rev a2)
...
```

GM108 doesn't do NVENC so I am CPU bound...

GSahiDarsini, 2020-02-18:
> I reverted back to zed sdk 2.8 (docker image 40e7577bf374) and corresponding zed-python-api (branch zedsdk_2.X) and all the problems went away. More disturbingly was that the entire reason for this [...]

Could you please let me know how you reverted back to zed sdk 2.8.

dustymugs, 2020-02-19:
@GSahiDarsini Reverting may not be best term. I just deployed StereoLab's 2.8 docker image and then used the zedsdk_2.X branch of zed-python-api. I had to update my python scripts as the API did change between 2.8 and 3.0

amaanda, 2020-02-27:
I'm using zed 2.8.3 for my application and, after recording a few files using lossless compression, when i try recording the next one, it simply won't save/start recording. any ideas?

closed 2022-04-27 by github-actions[bot]