stereolabs/zed-python-api #135: Failing to Record SVO on Jetson Nano

areiner222, 2020-02-18:
I cannot record an svo file on the Jetson Nano either via the python api or with the 'recording' sample (ZED_SVO_RECORDING).

I am working with: Jetson Nano with L4T 32.3.1 [ JetPack 4.3 ], CUDA 10.0.326, Kernel Version: 4.9.140-tegra, Python 3.6.8, ZED SDK JP4.3 v3.0.2 and ZED Python API built from up-to-date source.

When I run the SVO recording sample, I get the following :

> Opening zed camera was a SUCCESS
> Opening in BLOCKING MODE
> NvMMLiteOpen : Block : BlockType = 4
> ===== NVMEDIA: NVENC =====
> NvMMLiteBlockCreate : Block : BlockType = 4
> NvMapMemHandleAlloc: invalid size
> nvbuf_utils: Unable to allocate HW buffer for PayloadType MemHandle
> NvBufferCreateEx with memtag 4608 failed
> Encoder failed to allocate CAPTURE_PLANE buffer
> NvVideoEnc: NvVideoEncAllocateOutBuffers failed err=-1
> Aborted (core dumped)

The output svo file gets created but no data is recorded.

When I run the following python script:
```
import pyzed.sl as sl

zed = sl.Camera()

init_params = sl.InitParameters()
init_params.sdk_verbose = True

err = zed.open(init_params)
print("Opening zed camera was a", err)

runtime_parameters = sl.RuntimeParameters()
runtime_parameters.sensing_mode = sl.SENSING_MODE.STANDARD  # Use STANDARD sensing mode

video_name = 'test.svo'
params = sl.RecordingParameters(
    video_filename=video_name,
    compression_mode=sl.SVO_COMPRESSION_MODE.H264
)
err = zed.enable_recording(params)
# script never gets here
print("Enable recording msg", err)
zed.close()
```

I get the error

> [ZED][Init] Depth mode: PERFORMANCE
> [ZED][Init] Video mode: HD720@60
> Opening zed camera was a SUCCESS
> Opening in BLOCKING MODE
> NvMMLiteOpen : Block : BlockType = 4
> ===== NVMEDIA: NVENC =====
> NvMMLiteBlockCreate : Block : BlockType = 4
> Killed

obraun-sl, 2020-02-19:
Hi,

I tried to reproduce your issue without success (Nano, JP4.3)
- Can you try with C++ SVO recording sample to check if you have the same issue?
- Did you make sure that Tegra multimedia API was installed during jetpack installation ? Normally it's checked by default.

OB

areiner222, 2020-02-19:
Appreciate the help here.

- The C++ SVO recording sample is the sample in /usr/local/zed/sample/svo recording/recording, correct ? The result of building and running that is the first error in my previous post.
- I flashed the jetpack with defaults. Is there a way I can verify the tegra multimedia api ?

A few updates:
- I am using the original ZED camera (not ZED 2)
- I reinstalled the 2.8 SDK and built the python api for zedsdk_2.X from source, and now I am successfully able to record an SVO file with both the sample as well as through the api. Seems to be something in 3.0 that is causing my issue ?

obraun-sl, 2020-02-19:
Thanks for the Info. If it's working with 2.8, then the Jetpack is definetely  correct. 
I will try to reproduce with a ZED and let you know

obraun-sl, 2020-02-26:
Please update to 3.0.3. This should fix the issue.

closed 2020-06-09 by adujardin