stereolabs/zed-python-api #136: Problems with running tutorials on Jetson Xavier (Ubuntu 18.04)

GaneshArvapalli, 2020-02-19:
**Error:**
I've been having issues with both the Zed and the Zed 2 regarding an import error for pyzed.sl. The exact error for calling `python3 tutorials/tutorial\ 1\ -\ hello\ ZED/hello_zed.py` is as follows:
```
Traceback (most recent call last):
  File "tutorials/tutorial 1 - hello ZED/hello_zed.py", line 21, in <module>
    import pyzed.sl as sl
ImportError: libsl_core.so: cannot open shared object file: No such file or directory
```

**Software Versions:**
I'm on ZED SDK 3.0.2 with CUDA 10.0. My Python version is 3.6.

**Attempts at a solution:**
I've tried following the instructions from [this link](https://github.com/stereolabs/zed-python-api/issues/122), but I'm still getting this error, even after rebuilding/installing using `sudo python3 setup.py build`/`install` and removing the SDK 2.8 install completely.

The output of `ls /usr/local/lib/python3.6/dist-packages/pyzed*` is
```
/usr/local/lib/python3.6/dist-packages/pyzed-3.0.egg-info

/usr/local/lib/python3.6/dist-packages/pyzed:
__init__.py  __pycache__  sl.cpython-36m-aarch64-linux-gnu.so
```

Thanks for the help!

adujardin, 2020-02-19:
It seems there's still a 2.8 version of pyzed installed somewhere, could you try this https://github.com/stereolabs/zed-python-api/issues/122#issuecomment-579348683 ? 

Just in case you can remove (as in `rm -rf`) everything pyzed related, then reinstall the version compatible with the ZED SDK 3.0

GaneshArvapalli, 2020-02-19:
This was the solution. Running sudo updatedb again revealed that there were instances of pyzed-2.8 that had not been removed after the first updatedb/rm -rf. Thank you!

closed 2020-02-19 by GaneshArvapalli