stereolabs/zed-python-api #139: get_timestamp(sl.TIME_REFERENCE.IMAGE) returns CURRENT timestamp and not IMAGE timestamp

JesperChristensen89, 2020-02-23:
On Jetson Nano with newest SDK and Jetpack (downloaded installed this week). Works in C++.

Code snippet to reproduce:

```python
import pyzed.sl as sl
import numpy as np
import cv2

if __name__ == "__main__":
    path = "test.svo"

    init_params  = sl.InitParameters()
    init_params.input.set_from_svo_file(path)
    init_params.svo_real_time_mode = False
    init_params.depth_mode = sl.DEPTH_MODE.QUALITY
    init_params.coordinate_units = sl.UNIT.MILLIMETER  # Use milliliter units (for depth measurements)

    zed = sl.Camera()

    status = zed.open(init_params )
    if status == sl.ERROR_CODE.SUCCESS:
        print("Opened {}".format(path))
    else:
        print("Error: Could not open {}".format(path))

    # Get image size
    image_size = zed.get_camera_information().camera_resolution
    width = image_size.width
    height = image_size.height

    # Prepare single image containers
    left_image = sl.Mat()
    right_image = sl.Mat()
    depth_image = sl.Mat()
    depth_matrix = sl.Mat()

    rt_param = sl.RuntimeParameters()
    rt_param.sensing_mode = sl.SENSING_MODE.FILL

    nb_frames = zed.get_svo_number_of_frames() # doesn't work

    while True:
        if zed.grab(rt_param) == sl.ERROR_CODE.SUCCESS:
            svo_position = zed.get_svo_position() # doesn't work

            # Retrieve SVO images
            zed.retrieve_image(left_image, sl.VIEW.LEFT)
            zed.retrieve_image(right_image, sl.VIEW.RIGHT)
            zed.retrieve_image(depth_image, sl.VIEW.DEPTH)

            # retrive depth matrix
            zed.retrieve_measure(depth_matrix, sl.MEASURE.DEPTH)

            timestamp = zed.get_timestamp(sl.TIME_REFERENCE.IMAGE).get_seconds()
```

adujardin, 2020-03-03:
I tried to replicate this issue without success. Different settings, live or SVO and other platforms. I used the following snippet, basically identical to yours with a few more prints. 

From my tests : 
- the IMAGE ts are always identicals, meaning it's not the CURRENT values
- the CURRENT values are always different suggesting it's also functional

```python3
import pyzed.sl as sl
import numpy as np

if __name__ == "__main__":
    path = "test.svo"

    init_params  = sl.InitParameters()
    init_params.set_from_svo_file(path)
    init_params.svo_real_time_mode = False
    init_params.depth_mode = sl.DEPTH_MODE.QUALITY
    init_params.coordinate_units = sl.UNIT.MILLIMETER  # Use milliliter units (for depth measurements)

    zed = sl.Camera()

    status = zed.open(init_params)
    if status == sl.ERROR_CODE.SUCCESS:
        print("Opened {}".format(path))
    else:
        print("Error: Could not open {}".format(path))

    # Get image size
    image_size = zed.get_camera_information().camera_resolution
    width = image_size.width
    height = image_size.height

    # Prepare single image containers
    left_image = sl.Mat()
    right_image = sl.Mat()
    depth_image = sl.Mat()
    depth_matrix = sl.Mat()

    rt_param = sl.RuntimeParameters()
    rt_param.sensing_mode = sl.SENSING_MODE.FILL

    while True:
        if zed.grab(rt_param) == sl.ERROR_CODE.SUCCESS:

            # Retrieve SVO images
            zed.retrieve_image(left_image, sl.VIEW.LEFT)
            zed.retrieve_image(right_image, sl.VIEW.RIGHT)
            zed.retrieve_image(depth_image, sl.VIEW.DEPTH)

            # retrive depth matrix
            zed.retrieve_measure(depth_matrix, sl.MEASURE.DEPTH)

            timestamp_cur1 = zed.get_timestamp(sl.TIME_REFERENCE.CURRENT).get_microseconds()
            timestamp_im1 = zed.get_timestamp(sl.TIME_REFERENCE.IMAGE).get_microseconds()
            timestamp_im2 = zed.get_timestamp(sl.TIME_REFERENCE.IMAGE).get_microseconds()
            timestamp_cur2 = zed.get_timestamp(sl.TIME_REFERENCE.CURRENT).get_microseconds()

            print(str(timestamp_im1) + " " + str(timestamp_im2) + " / " + str(timestamp_cur1) + " " + str(timestamp_cur2))
```

JesperChristensen89, 2020-03-03:
We can close it. I haven't tried this again. In the meantime I updated to 3.0.3 and switched to C++. No issues there.

closed 2020-03-03 by JesperChristensen89