stereolabs/zed-python-api #140: Recording does not work on Jetson platform

JesperChristensen89, 2020-02-24:
On Jetson TX2, newest Jetpack, newest SDK (downloaded and installed last week). 
Code snippet to reproduce:

```python
import pyzed.sl as sl
import argparse
import cv2
import time

ap = argparse.ArgumentParser()
ap.add_argument("-i", "--ip", type=str, required=True,
    help="IP to the sender of the stream")
ap.add_argument("-p", "--port", type=int, required=True,
    help="port used for receiving the stream")

args = vars(ap.parse_args())

if __name__ == "__main__":

    # Create a ZED camera object
    zed = sl.Camera()

    # Set configuration parameters
    init_params = sl.InitParameters()
    init_params.depth_mode = sl.DEPTH_MODE.NONE
    init_params.set_from_stream(args["ip"],args["port"])
    init_params.sdk_verbose = False

    # Open the camera
    err = zed.open(init_params)
    while err != sl.ERROR_CODE.SUCCESS:
        print("Opening camera error: ", err,flush=True)
        print("Trying again in 1 second ...",flush=True)
        time.sleep(1)
        err = zed.open(init_params)

    print("Camera opened. Starting stream ...",flush=True)

    # Print camera info
    camera_info = zed.get_camera_information()
    print("\n",flush=True)
    print("ZED Model \t\t: {}".format(camera_info.camera_model),flush=True)
    print("ZED Serial Number \t: {}".format(camera_info.serial_number),flush=True)
    print("ZED Camera Firmware \t: {}-{}".format(camera_info.camera_firmware_version, camera_info.sensors_firmware_version),flush=True)
    print("ZED Camera Resolution \t: {}x{}".format(camera_info.camera_resolution.width, camera_info.camera_resolution.height),flush=True)
    print("ZED Camera FPS \t\t: {}".format(camera_info.camera_fps),flush=True)

    # Enable recording
    recording_param = sl.RecordingParameters("./test.svo",sl.SVO_COMPRESSION_MODE.H264)
    err = zed.enable_recording(recording_param)
    if err != sl.ERROR_CODE.SUCCESS:
        print(repr(status))
        exit(1)

    runtime = sl.RuntimeParameters()
    runtime.enable_depth = False

    mat = sl.Mat()

    key = ''
    while key != 113:  # for 'q' key
        err = zed.grab(runtime)
        if err == sl.ERROR_CODE.SUCCESS:
            zed.retrieve_image(mat, sl.VIEW.LEFT)
            cv2.imshow("ZED", mat.get_data())
        key = cv2.waitKey(5)

    cv2.destroyAllWindows()
    zed.disable_recording()
    zed.close()
```
It seems it floods the memory and then exits. Errors are not really consistent. Here's one example of an error:
```
Opening in BLOCKING MODE                                                         
NvMMLiteOpen : Block : BlockType = 261                                           
NVMEDIA: Reading vendor.tegra.display-size : status: 6                           
NvMMLiteBlockCreate : Block : BlockType = 261                                   
libv4l2_nvvidconv (0):(802) (INFO) : Allocating (11) OUTPUT PLANE BUFFERS Layout =1
libv4l2_nvvidconv (0):(818) (INFO) : Allocating (11) CAPTURE PLANE BUFFERS Layou t=0
Camera opened. Starting stream ...                                                                                                                                                                                                                 
ZED Model               : ZED-M                                                  
ZED Serial Number       : 12811826                                               
ZED Camera Firmware     : 1523-517                                               
ZED Camera Resolution   : 1280x720                                               
ZED Camera FPS          : 60.0                                                   
Opening in BLOCKING MODE                                                         
NvMMLiteOpen : Block : BlockType = 4                                             
===== NVMEDIA: NVENC =====                                                       
NvMMLiteBlockCreate : Block : BlockType = 4                                      
NvMapMemHandleCreate: error 22                                                   
nvbuf_utils: Unable to allocate HW buffer for PayloadType SurfArray              
NvBufferCreateEx with memtag 4608 failed                                         
nvbuf_utils: dmabuf_fd 0 mapped entry NOT found                                  
nvbuf_utils: Can not get HW buffer from FD... Exiting...                         
Segmentation fault (core dumped)
--
```
If I comment out the `enable_recording()` lines, it works just fine.

adujardin, 2020-02-25:
We've identified the problem, it will be fixed in an upcoming patch. 
Sorry for the inconvenience and thank you for reporting the issue.

adujardin, 2020-02-25:
The ZED SDK 3.0.3 is now released and includes this fix https://www.stereolabs.com/developers/release/

areiner222, 2020-02-26:
@adujardin can you please comment whether the newest patch will fix my issue as well ? https://github.com/stereolabs/zed-python-api/issues/135

closed 2020-06-09 by adujardin