stereolabs/zed-python-api #145: Zed Depth Viewer Point Cloud Only Available in Standard Mode

saratrajput, 2020-04-24:
Why does `Zed Depth Viewer` show point cloud is only available in `Standard Mode`? Is it supposed to be like this or am I missing something here? 
[image]

I'm using `Zed SDK 3.1`, with `Zed-M` camera on `Ubuntu 18`.

obraun-sl, 2020-04-24:
Hi,

This is the expected behavior since we want to indicate that Point cloud visualization gives better results in the STANDARD mode  (because it's more precise, less filtering). Therefore, in **Depth Viewer**, we don't show the point cloud in FILL mode.
However, using the ZED SDK, you can of course access the point cloud Mat using FILL or STANDARD mode without restrictions.

saratrajput, 2020-04-27:
@obraun-sl Thanks for the quick reply. I just had a quick follow up question.
So is it advisable to use `STANDARD` mode with `confidence_threshold` and `textureness_confidence_threshold` at 100 or `FILL` mode for applications where accuracy of the point cloud is more important? I couldn't understand from the documentation itself.

obraun-sl, 2020-04-27:
If accuracy of the point cloud is the most important, always use the STANDARD mode. 
To increase/decrease density of the depth map, you can change the confidence/texture confidence, depending on the level of density you need.

saratrajput, 2020-04-27:
Perfect. Thank you.
And one last thing I wanted to clear. About `QUALITY` and `ULTRA` mode. Is ULTRA mode only meant to be used when the objects are farther away? Or should I expect better results when objects are within 0.3 ~ 1m in well lit environment from ULTRA mode than QUALITY mode? Processing time  and memory not being an issue.

saratrajput, 2020-04-28:
Anyway I'll close this issue since my initial question was answered.

closed 2020-04-28 by saratrajput