stereolabs/zed-python-api #147: ImportError

donMarine, 2020-06-04:
Hi,

I am on Windows 10 with Cuda 10.2, ZED SDK 3.1.2 and python 3.8.1. I have installed the pyzed API and am able to import pyzed but when I attempt to import pyzed.sl I get the error:
`ImportError: DLL load failed while importing sl: The specified module could not be found.`
I have checked the module folder and see the sl.pyd file.
Has anyone else experience the same issue and how can I fix it?

Thanks

eric-vader, 2020-06-05:
Hi I have the same exact error message. I have been banging my head trying to solve this...

cfrend, 2020-06-06:
Same issue here, I tried this solution and it did not work https://github.com/stereolabs/zed-python-api/issues/78 in manually building/installing the pyzed api.  As well as rolling back to CUDA 10 and ZED sdk for CUDA 10 based on this link https://github.com/stereolabs/zed-python-api/issues/69 and I still cannot import pyzed.sl in python 3.8 64bit.  I have been trying to solve on Windows 10 for the last 5 hours with no luck.

`Traceback (most recent call last):
  File ".\hello_zed.py", line 21, in <module>
    import pyzed.sl as sl
ImportError: DLL load failed while importing sl: The specified module could not be found.`

RomanBats, 2020-06-28:
Same issue.
@nesnes, @adujardin, @obraun-sl, @P-yver any suggestions?

noelquah, 2020-07-02:
same issue here too.

mjlyons7, 2020-07-13:
@noelquah Same with me, it worked first try with my desktop, but I keep getting the dll import error on my laptop. Laptop is using MX150 GPU, desktop is GTX 1070. All software settings (CUDA, ZED SDK, Python) were the same. Maybe there is a bug with the Python API with laptops? That is my only guess, after trying lots of different software combinations to try to get this to work with my laptop.

noelquah, 2020-07-13:
@mjlyons7 I actually fixed the error on my laptop but i did a couple of things and I can't pinpoint the actual cause. 

1. I downgraded my Python from 3.8 to 3.6. I noticed that it was the difference between my 2 devices. This solution did stop this error from occurring but then another error came up saying that pyzed was not found (no longer sl).

2. I made a silly mistake by linking my virtual environment to my IDE - at the time, my VE's python had no pyzed installed. I linked it to the installed version and then it worked fine. 

I don't know which item precedes the other, but maybe you could give these solutions a go.

kyelewis, 2020-07-21:
Same problem here and @noelquah - installing Python 3.6 for me resolved the issue.

Does not work:
Python 3.8.5 (tags/v3.8.5:580fbb0, Jul 20 2020, 15:57:54) [MSC v.1924 64 bit (AMD64)] on win32

Works fine:
Python 3.6.4 (v3.6.4:d48eceb, Dec 19 2017, 06:54:40) [MSC v.1900 64 bit (AMD64)] on win32

A7ocin, 2020-08-07:
I confirm that the Python APIs are not working with Python 3.8. When will this be fixed?

A7ocin, 2020-09-23:
I solved the issue by copying all the dlls from
**C:\Program Files (x86)\ZED SDK\bin**
into Python's Script folder

xu-fire, 2020-12-07:
It can‘t  work with Python3.9.0.I get the error:
ImportError: DLL load failed while importing sl: The specified module could not be found.

A7ocin, 2020-12-08:
@xu-fire try the solution I suggested in my latest reply, it should work

cbayazit, 2021-01-01:
Running on Windows. 10 , I have Cuda 11.1 and Python 3.8.6 , no luck importing pyzed.sl. Any help is much appreciated.

EliVerbrugge, 2021-01-10:
Hmm, I am also encountering this issue with Python 3.8.6, ZED SDK 3.3 and CUDA 11.1 on Windows. I tried to copy the dlls in C:\Program Files (x86)\ZED SDK\bin to my venvs script folder as @A7ocin suggested, but that did not seem to fix my issue. Maybe I am missing something.

kunalkumthekar, 2021-02-17:
Hi,
im on zed sdk 3.4, cuda 11.1, python 3.9 . im neither able to install the api directly by running the get_python_api file nor able to install it manually properly. When I try to install/build api manually from the source file, i get the "Unable to find vcvarsall.bat" error. Please can someone help me with the issue?
Regards

qt-truong, 2021-02-17:
Hi @kunalkumthekar,

There might be some issues with Python versions > 3.8. 
If it is possible for you, could you try reinstalling the API with a version < 3.8 ?

kunalkumthekar, 2021-02-17:
@qt-truong
Thank you for the quick response. Yes I did try running the entire procedure on python 3.6 as suggested above. Still no luck. Is the api installation procedure going wrong for me? Can you please tell me how you resolved this issue?

qt-truong, 2021-02-17:
@kunalkumthekar it seems that you have an issue with your C++ compiler.
Can you try following the suggestions on [this page](https://devblogs.microsoft.com/python/unable-to-find-vcvarsall-bat/) and see if that solved your problem ?

kunalkumthekar, 2021-02-17:
@qt-truong Hey thank you for the help, I was able to solve the issue by updating Visual Studio to a newer version. Also, I had issues with the python version 3.9 , 3.8 so I downgraded it to 3.6 as suggested by you and others.
Regards!

GSORF, 2021-05-30:
I can confirm the error

`Traceback (most recent call last):
  File "C:\Program Files (x86)\ZED SDK\samples\svo recording\playback\python\read_svo.py", line 26, in <module>
    import pyzed.sl as sl
ImportError: DLL load failed while importing sl: Das angegebene Modul wurde nicht gefunden.`

when running

`python "C:\Program Files (x86)\ZED SDK\samples\svo recording\playback\python\read_svo.py" {FILEPATH_TO_MY_SVO_RECORDING}\HD1080_SN26393711_03-53-17.svo`
on the following system: _Windows 10, Python 3.8.3 (tags/v3.8.3:6f8c832, May 13 2020, 22:37:02) [MSC v.1924 64 bit (AMD64)] on win32_

Thanks to @EliVerbrugge and @A7ocin I have copied the two DLL files "sl_ai64.dll" and "sl_zed64.dll" from the path "C:\Program Files (x86)\ZED SDK\bin" to the location at "C:\Program Files\Python38\Lib\site-packages\pyzed". After this has been done, reading the SVO file and displaying it using OpenCV works fine.

How can we contribute to fix this issue?

Kind regards,
Adam

CircleOnCircles, 2021-08-06:
> Hmm, I am also encountering this issue with Python 3.8.6, ZED SDK 3.3 and CUDA 11.1 on Windows. I tried to copy the dlls in C:\Program Files (x86)\ZED SDK\bin to my venvs script folder as @A7ocin suggested, but that did not seem to fix my issue. Maybe I am missing something.
> 
> EDIT: I resolved this issue by copying the dlls in the bin folder to the Python\Lib\site-packages\pyzed folder. Thanks @A7ocin.

run `python -m site` locate the path and copy from C:\Program Files (x86)\ZED SDK\bin to site-packages\pyzed solved the issue.

im using python  3.8.9

closed 2022-04-27 by github-actions[bot]

P-yver, 2022-05-19:
Indeed this behavior can append with various python version (in my case it works with 3.9 but I get the error with 3.10).

The solution pointed multipled times to copy the ZED SDK DLL into the python directory is the best, [as mentionned](https://github.com/stereolabs/zed-python-api/issues/147#issuecomment-894377297)

KainanSu, 2022-06-08:
> 

I use conda, and this `Python Path` can still help it work.
Thanks.

hkravn, 2022-06-30:
I am using ZED SDK 3.6.5 with Python 3.10.2 and CUDA 11.5
Even after copying the dlls mentioned from BIN folder to pyzed site-packages, the issue is not being resolved.
Someone please help to resolve the issue

akawincent, 2023-05-09:
@hkravn 
There is an issue that identical with yours confusing me.
I am also using Python 3.10.2.
I think this issue perhaps has to do with the version of Python.

akawincent, 2023-05-10:
@xu-fire 
Thanks for your useful recommendation. 
What's more, I found that reinstall ZED SDK also works within Python 3.10.2.

ShanMallinathan, 2023-06-08:
Hi,
I have tried SDK 4.0 with Python 3.8.6 and could not get rid of this error despite copying the DLL files. Would someone be able to help me resolve this? Thanks in advance!

akawincent, 2023-06-09:
@ShanMallinathan 
I have tried to reinstall the SDK 3.8 with Python 3.10.2, instead of SDK 4.0.
I suggest you lower the version of SDK. This method can resolve the annoying question.
Hope this advice is useful to you！

ShanMallinathan, 2023-06-15:
Thanks @akawincent 
I am not too sure why that did not work for me. But tried to install it in my Linux and it works without any problem

akawincent, 2023-06-16:
> Thanks @akawincent I am not too sure why that did not work for me. But tried to install it in my Linux and it works without any problem

Perhaps something went wrong when you ran 'get_python_api.py'.

MMertBedez, 2024-05-25:
> I solved the issue by copying all the dlls from **C:\Program Files (x86)\ZED SDK\bin** into Python's Script folder

Bro where is the "Python Script Folder"??

ASRodrigo1, 2024-07-08:
> I solved the issue by copying all the dlls from **C:\Program Files (x86)\ZED SDK\bin** into Python's Script folder

Worked for me as well, using miniconda.
Python 3.9
ZED SDK 4.0.1
CUDA 11.8
Windows 11

I just copied all .dll files from **C:\Program Files (x86)\ZED SDK\bin** to 
1 - **C:\Users\user_name\.conda\envs\zed\Lib\site-packages\pyzed**
2 - **C:\Users\user_name\.conda\envs\zed\Lib\site-packages**