stereolabs/zed-python-api #148: supporting ZED SDK 3.2 (CUDA 10.2,SDK 3.2)

ryota2425, 2020-06-11:
Im using Jetson xavier NX.
Will you provide support for SDK 3.2?
Thank you.

harshilpatel, 2020-06-12:
same for CUDA 9.0 SDK < 3

adujardin, 2020-06-12:
The NX on 3.2 is supported, there was an issue in the script packaged in the installer for the JP44 at the release, but it's now fixed.

You can download this script to get the whl file. All recent supported platform should be handled :

```sh
# Redownload the script just in case, but should already be present
wget download.stereolabs.com/zedsdk/pyzed -O /usr/local/zed/get_python_api.py
python3 /usr/local/zed/get_python_api.py
```

@harshilpatel For older 2.X version you should check this release https://github.com/stereolabs/zed-python-api/releases/tag/v2.8.x

ryota2425, 2020-06-12:
I redownloaded this file and it all worked all. Appreciate the help here.

closed 2020-06-12 by ryota2425