stereolabs/zed-python-api #149: Depth sensing in FILL MODE does not work on Jetson Xavier (zed mini)

dige-dev, 2020-06-16:
Hello,

It seems like Depth sensing in FILL MODE does not work on Jetson Xavier. I get a blank image when i try to display depth, please advise.

solimov-advanced, 2020-09-01:
I'm also experiencing this issue, started happening suddenly after updating to ZED SDK v3.1 and v3.2 on multiple Jetson Xaviers with JetPack4.3. Reverting to ZED SDK v3.0 seems to work. (this was checked using the ZED Depth Viewer tool on FILL mode, SDK 3.1 and 3.2 return a black image :((( )

Rylynn1209, 2021-01-25:
I'm having the same problem on ZED2 and Jetson Xavier NX with JetPack 4.4.1 + CUDA 10.2 + ZED SDK 3.3.3(for JetPack 4.4). STANDARD mode works, but in FILL mode the depth is a black image.

closed 2022-04-27 by github-actions[bot]