stereolabs/zed-python-api #153: ERROR: pyzed-3.1-cp37-cp37m-linux_x86_64.whl is not a supported wheel on this platform.

HassanB93, 2020-09-22:
Hi

I am trying following code on the Ubuntu 20.04 (python 3.8.2) based in your instruction to install "pyzed-3.1" package. 

python3 -m pip install pyzed-3.1-cp37-cp37m-linux_x86_64.whl

However, I received this error:

WARNING: Requirement 'pyzed-3.1-cp37-cp37m-linux_x86_64.whl' looks like a filename, but the file does not exist
ERROR: pyzed-3.1-cp37-cp37m-linux_x86_64.whl is not a supported wheel on this platform.

I would appreciate if you help me on this issue.

adujardin, 2020-09-22:
Hi

- The ZED SDK 3.1 doesn't support Ubuntu 20, you need the ZED SDK 3.2.
- You're trying to install the python 3.7 wheel package using python 3.8, the versions have to match

You need to install the C++ SDK for [Ubuntu 20 (CUDA 11.0)](https://download.stereolabs.com/zedsdk/3.2/cu110/ubuntu20) first, then run :

```
python3 /usr/local/zed/get_python_api.py
```
It will download the correct version for your platform and SDK version

HassanB93, 2020-09-23:
Thank you @adujardin 

Is it possible to work with the ZED camera on a machine without NVIDIA graphic card. I mean getting depth map and point cloud from the camera on a machine with Mesa Intel® Iris(R) Plus Graphics 640.

adujardin, 2020-09-23:
The ZED SDK requires CUDA to run all functions.

However, you could use this sample https://github.com/stereolabs/zed-opencv-native that provides basic features using OpenCV. It rectifies the images using the factory calibration, then you can use OpenCV BM or SGBM to compute the depth map (you need to implement this part yourself).

LeoReubelt, 2021-08-13:
I am having the same problem using Windows 10 and python 3.7.11.  Any idea what could be causing it?

closed 2022-04-27 by github-actions[bot]