stereolabs/zed-python-api #155: 3.2 SDK has some invalid function

PonyPC, 2020-10-07:
```
	transform = sl.Transform()
	transform[0,0] = 0
	transform[0,1] = 1
	print(transform.get_rotation_matrix())
```
transform is always identity matrix no matter how mat is.

```
	tracking_params = sl.PositionalTrackingParameters(transform)
	transform2 = sl.Transform()
	print(tracking_params.initial_world_transform(transform2))
	print(transform2)
```
transform2 is always identity matrix no matter how transform is.

Others like get_orientation, get_euler_angles also.
Do I have mistake?

qt-truong, 2020-10-07:
Hi there,

You are trying to modify a Transform object, which contains rotation and translation data. By default, the rotation will be an identity matrix. If you want to set a custom rotation matrix to your transform object, you should try to do instead :

```python
transform = sl.Transform()

py_rotation = sl.Rotation()   # Identity matrix 3x3
py_rotation[0,0] = 0
py_rotation[0,1] = 1 

transform.set_rotation_matrix(py_rotation)
print(transform.get_rotation_matrix())
```
It would result in something like :

```python
0.000000 1.000000 0.000000
0.000000 1.000000 0.000000
0.000000 0.000000 1.000000
```
You should get the corresponding orientation and euler angles If you try `get_orientation()` and `get_euler_angles()` on the newly set `transform`.

As for transform2, you are also instantiating it by default so it will result in an identity matrix.

Hope it helps !

PonyPC, 2020-10-07:
I follow your code, it does print 
```
0.000000 1.000000 0.000000
0.000000 1.000000 0.000000
0.000000 0.000000 1.000000
```
But print(transform) is still an identity matrix.
```
1.000000 0.000000 0.000000 0.000000
0.000000 1.000000 0.000000 0.000000
0.000000 0.000000 1.000000 0.000000
0.000000 0.000000 0.000000 1.000000
```
If I set 
```
tracking_params = sl.PositionalTrackingParameters(transform)
```
Is tracking_params initialized by identity matrix or not?

closed 2020-10-15 by PonyPC