stereolabs/zed-python-api #156: Can't install zed python api

chaoren2357, 2020-10-18:
ZED SDK Version: 3.2.2
Platform win
Python 3.8

I follow the steps from website https://www.stereolabs.com/docs/app-development/python/install/#installing-the-python-api
The output shows:
```
(zed_env) C:\Program Files (x86)\ZED SDK>python get_python_api.py
Platform win
ZED 3.2
Python 3.8
Downloading python package from https://download.stereolabs.com/zedsdk/3.2/win//py38 ...

Unsupported platforms, no pyzed file available for this configuration
 It can be manually installed from source https://github.com/stereolabs/zed-python-api
```
But I can't find the ZED 3.2 release version in this repo.  What should I do to install those packages?

adujardin, 2020-10-19:
There was an issue on Windows with CUDA 11, you can try to download the updated script from [here](https://download.stereolabs.com/zedsdk/pyzed), it should fix the issue.

Other users have reported some problems with Python 3.8 on Windows, but this is different from this (the root cause is not clearly identified yet).

chaoren2357, 2020-10-19:
> There was an issue on Windows with CUDA 11, you can try to download the updated script from [here](https://download.stereolabs.com/zedsdk/pyzed), it should fix the issue.
>
> Other users have reported [...]

I have tried the script, but it doesn't work. The messages are as follows:
```
(zed_env) C:\Program Files (x86)\ZED SDK>python get_pyzed.py
CUDA 11.1
Platform win
ZED 3.2
Python 3.8
Downloading python package from https://download.stereolabs.com/zedsdk/3.2/win/cu111/py38 ...

Unsupported platforms, no pyzed file available for this configuration
 It can be manually installed from source https://github.com/stereolabs/zed-python-api

```
Is there any more information I need to provide to solve this issue?

adujardin, 2020-10-19:
Ok now the detection works but your platform is actually not supported, unfortunately. 
You need CUDA 11.0 and not 11.1, the minor version must also match the requirements. 

We can't release a ZED SDK for CUDA 11.1 until Nvidia provides the compatible AI libraries (TensorRT).

chaoren2357, 2020-10-20:
> Ok now the detection works but your platform is actually not supported, unfortunately. 
> You need CUDA 11.0 and not 11.1, the minor version must also match the requirements. 
> 
> We can't release a ZED SDK for CUDA 11.1 until Nvidia provides the compatible AI libraries (TensorRT).

Thanks for your help.  I will try cuda 11.0.

closed 2020-10-20 by chaoren2357