stereolabs/zed-python-api #162: get_device_list() returns nothing

JosefGansinger, 2020-11-24:
return sl.Camera.get_device_list()

returns empty list. Anyone else has this Problem?

qt-truong, 2020-11-25:
Hi @JosefGansinger,

This is very odd indeed. Have you checked that your camera is well plugged in a USB 3 port and well detected by your PC ?

Helaly96, 2020-11-25:
if it's connected to a port that is not a USB3 it will do that

JosefGansinger, 2020-11-25:
Setup is: JetsonNX + 1x ZED2

@qt-truong yes. the device is already found. - but the sl.Camera.get_device_list() function doesn't list any device
@Helaly96 and it is connected via USB3 - on a JastonNX

I think i found the issue - when the device is already open, it won't show up in the available devices any more. 
Please close.
thank you

closed 2020-11-26 by JosefGansinger