stereolabs/zed-python-api #163: Segmentation fault when using body tracking on Jetson Xavier NX

mcwl, 2020-12-10:
Hello,

Each time I try to use body tracking via Python API on Jetson Xavier NX I get Segmentation fault error.

It happens when using code from Samples and in my own script. The C++ example works perfect.

The board has the newest versions of L4T, Jetpack (4.4) and ZED SDK (3.3.3) installed with all default options without any errors.

Any ideas how to resolve this?

ben-mcdonnell-hacksmith, 2020-12-27:
Hey mcwl,
I am having the exact same issue, I also have been working on the same project involved with #161, and the root problem is the same, it just wasn't diagnosed properly in the beginning.  

This does seem like an issue with the ZED API, so hopefully the devs can get on this sooner than later.

qt-truong, 2021-01-14:
Hi, 

The [ZED SDK 3.4 RC](https://community.stereolabs.com/t/zed-sdk-3-4-0-release-candidate/58) should solve this issue

closed 2022-04-27 by github-actions[bot]