stereolabs/zed-python-api #164: Illegal instruction (core dumped)

Shishir-Kumar-Singh, 2021-01-10:
I am getting this error at the time of importing pyzed library in python sample program..
python positional_tracking.py 
Illegal instruction (core dumped)

SDK version : 3.3
Device: Jetson with jetpack 4.4

closed 2021-01-12 by Shishir-Kumar-Singh

raman123456789, 2021-04-20:
i am aswell getting the same error, could you please share how did you solved it.

Shishir-Kumar-Singh, 2021-04-21:
@raman123456789 What is your configuration...can you elaborate please like SDK version, OS and the exact problem scenario because this issue come up generally because of lib issues.

qt-truong, 2021-04-21:
Hi @raman123456789,

Can you check [this answer](https://github.com/stereolabs/zed-python-api/issues/167#issuecomment-784050030) and see if it helps with your issue ?

wasdee, 2022-08-08:
> Hi @raman123456789,
> 
> Can you check [this answer](https://github.com/stereolabs/zed-python-api/issues/167#issuecomment-784050030) and see if it helps with your issue ?

this is working for me.

Jetpack 4.6.1