stereolabs/zed-python-api #165: AttributeError: type object 'pyzed.sl.Translation' has no attribute 'dot_translation'

Shishir-Kumar-Singh, 2021-01-10:
Getting this error while running the ZED sample python program on Jetson AGX Xavier
python3 positional_tracking.py 
Traceback (most recent call last):
  File "positional_tracking.py", line 50, in <module>
    viewer = gl.GLViewer()
  File "ZED/zed-examples-master/positional-tracking/python/ogl_viewer/tracking_viewer.py", line 162, in __init__
    self.camera = CameraGL()
  File "ZED/zed-examples-master/positional-tracking/python/ogl_viewer/tracking_viewer.py", line 476, in __init__
    self.setDirection(tmp, tmp2)
  File "ZED/zed-examples-master/positional-tracking/python/ogl_viewer/tracking_viewer.py", line 528, in setDirection
    if(sl.Translation.dot_translation(self.vertical_, self.up_) < 0.):
AttributeError: type object 'pyzed.sl.Translation' has no attribute 'dot_translation'

Platform: Jetson AGX Xavier, 
Jetpack 4.4
ZED SDK 3.3

qt-truong, 2021-01-11:
Hi @Shishir-Kumar-Singh,

The sample you launched uses a function that is not implemented in the SDK version you downloaded.
Can you try again with the SDK 3.4.0 RC1, available here : https://community.stereolabs.com/t/zed-sdk-3-4-0-release-candidate/58

Shishir-Kumar-Singh, 2021-01-12:
Thanks for the suggestion. It worked for me after installing 3.4.0 version.

closed 2021-01-12 by Shishir-Kumar-Singh