stereolabs/zed-python-api #167: illegal instruction zed-python-api on jetpack 4.5, sdk 3.4, Xavier AGX

bblumberg, 2021-02-06:
Hi
I have followed the installation instructions and consistently get illegal instruction error with the python samples and tutorials. This happens both in the directly installed sdk as well as the various docker images.

Does this need to be updated for either sdk 3.4 or Jetpack 4.5?

The c++ examples and tools all work fine.

Thanks

bb

adujardin, 2021-02-08:
Hi
Could you post more details about this error (a screenshot, the exact message, and configuration) and a simple case to reproduce it, maybe the docker image and commands used?

The only thing that came to mind with this message and mixing aarch64 and x86 docker images or maybe installers (this shouldn't be possible). It may be on our side during the compilation of the cython wrapper but this error was not reported before.

bblumberg, 2021-02-08:
Hi

Sorry not to provide more details. I am running on an Xavier AGX with
Jetpack 4.5. The error occurs both within docker (which I will walk though
below) and if I do a direct install. I assumed that it was an issue with
Jetpack 4.5 since the only docker image is for jp4.4, but as I said, I also
saw this when I did a direct install and followed the same steps in a clean
python3 environment. Thanks for your help!!!

bb

Reproduce on Docker

Docker command
docker run -it --runtime nvidia --privileged -e DISPLAY --network host
--device /dev/video0 -v /tmp/.X11-unix:/tmp/.X11-unix
stereolabs/zed:3.3-tools-devel-jetson-jp4.4

 Install necessary python libraries etc.
1. apt-get update
2. apt-get install python3-pip
3. pip3 install --upgrade pip
4. python3 -m pip install cython numpy opencv-python pyopengl
5. python3 get_python_api.py

Try the first tutorial:

root@bruce-ai:/usr/local/zed/samples/tutorials/tutorial 1 - hello
ZED/python# python3 hello_zed.py
Illegal instruction (core dumped)

I have attached the output from the terminal for your reference as well.

bblumberg, 2021-02-09:
As a test, I tried building zed-python-api within the docker container, and got the illegal instruction as well:

root@bruce-ai:/zed-python-api/src# python --version
Python 2.7.17
root@bruce-ai:/zed-python-api/src# python3 --version
Python 3.6.9
root@bruce-ai:/zed-python-api/src# python3 -m pip install cython numpy
Requirement already satisfied: cython in /usr/local/lib/python3.6/dist-packages (0.29.21)
Requirement already satisfied: numpy in /usr/local/lib/python3.6/dist-packages (1.19.5)
root@bruce-ai:/zed-python-api/src# python -c "import platform; print(platform.architecture())"
('64bit', '')
root@bruce-ai:/zed-python-api/src# python3 setup.py build
Illegal instruction (core dumped)

adujardin, 2021-02-09:
Ok thank you for all the information, that's very helpful. We'll look into it and try to reproduce and find a solution. We'll get back to you as soon as we have news.

OAkyildiz, 2021-02-15:
I got the same problem with Jetpak 4.4 and ZED SDK 3.4

```
    Python 3.6
    ZED SDK 3.4
```

bblumberg, 2021-02-22:
Sorry to bug you, but is there any chance that someone could look into this? This is a blocker for anyone who wants to use python and the sdk on the jetson family. Thanks.

adujardin, 2021-02-23:
Thank you for your patience, we tried different combinations and it seems the python binary of the system doesn't have this issue, only when using virtual env or conda somehow.

Since even compiling the Cython project trigger the error we can rule out our CI/packaging or an error within the code. From our testing, this fixes https://forums.developer.nvidia.com/t/illegal-instruction-core-dumped-xavier/166278/4 (`OPENBLAS_CORETYPE=ARMV8` before the python executable) found by another user on nvidia forum seems to solve the problem.

Could you test it and see if it also fixes the issue on your side?

bblumberg, 2021-02-23:
The workaround did fix the problem, and when I am running natively (outside
of docker) I am using virtualenvwrapper. So, BIG THANKS!

I also tried running on the stereolabs/zed:3.3-tools-devel-jetson-jp4.4
docker image, and it fixes the problem there, but I am getting other errors
there (for example, running the depth sensing example, I get:

File "depth_sensing.py", line 49, in <module>
    viewer = gl.GLViewer()
  File "/zed-lab/depth sensing/python/ogl_viewer/viewer.py", line 218, in
__init__
    self.camera = CameraGL()
  File "/zed-lab/depth sensing/python/ogl_viewer/viewer.py", line 459, in
__init__
    self.setDirection(tmp, tmp2)
  File "/zed-lab/depth sensing/python/ogl_viewer/viewer.py", line 513, in
setDirection
    if(sl.Translation.dot_translation(self.vertical_, self.up_) < 0.):
AttributeError: type object 'pyzed.sl.Translation' has no attribute
'dot_translation'
root@bruce-ai:/zed-lab/depth sensing/python# python3 -m pip install pyopengl
Requirement already satisfied: pyopengl in
/usr/local/lib/python3.6/dist-packages

When I tried stereolabs/zed:3.3-py-devel-jetson-jp4.4 docker image, I
didn't need to use the OPENBLAS... work-around but got the same error as
above. But that is for a different ticket I guess.

Thanks again for your help. It is much appreciated.

adujardin, 2021-02-24:
Glad to hear this workaround is fixing the problem. Openblas is used by the ZED SDK so I guess it's required for some version of the package when using the SDK.

The missing "dot_translation" is our fault, that's because the docker images contain the ZED SDK 3.3.0 and the python wrapper was improved in later patches (3.3.1 and 3.3.2). The python samples with OpenGL display are expecting a ZED SDK >= 3.3.1
I'll fix this by regenerating the 3.3 docker images.

closed 2022-04-27 by github-actions[bot]