stereolabs/zed-python-api #174: CAMERA STREAM FAILED TO START

kshamaramesh, 2021-03-24:
Hello,

I am just trying to run the sample code of hellozed.py but my system is not detecting the camera 

There is no issues with import of pyzed but I could not stream the camera.

Any help would be appreciated

qt-truong, 2021-03-24:
Hi @kshamaramesh,

The first tutorial is only printing your camera serial number in the terminal, you are not supposed to see the camera stream.
If that is indeed the case, there is nothing wrong with your camera and you can test other tutorials or samples.
Otherwise, you can try to open ZED_Explorer tool to see if your camera is detected or not by your system

kshamaramesh, 2021-03-24:
Thanks for the feedback!

It is working with the ZED_Explorer.

But next I tried to run the code of Image capture code example..

I ended up in **NO GPU COMPATIBLE**

My Graphics card is **NVIDIA  GeFORCE GTX 680**
CUDA Version **11.0**
**ZED SDK is 3.4.3 the latest version**

Please do let me know what could do I need to install the different version of SDK.

Thanks in advance

closed 2022-04-27 by github-actions[bot]

fmarcantoni, 2025-02-27:
Hello @qt-truong and @kshamaramesh,
I'm following up on this issue as I've been trying to run the image capture example with pyzed, but the python program does not detect the ZED Mini camera. I tried running the ZED Explorer and ZED Diagnostics, and they both seem to have no problem. If you were able to figure out this issue, let me know whenever you get the chance. Thank you!