stereolabs/zed-python-api #176: Compass heading

Bovakinn, 2021-03-26:
Hi, 
Are there any methods in the API to get the current compass heading relative to magnetic north?

Myzhar, 2021-03-26:
@Bovakinn there is currently a bug in Magnetometer Calibration that does not allow to properly calculate a valid heading using calibrated magnetic field values.
The fix will be released soon with SDK v3.5
We will also publish a tutorial about how to calculate the heading, meanwhile, you can read [this document](https://www.researchgate.net/publication/280112308_Measurement_and_Estimation_of_3D_Orientation_using_Magnetic_and_Inertial_Sensors)

Bovakinn, 2021-03-26:
@Myzhar That's brilliant thank you. 
Is there any timeframe available for v3.5 and the tutorial?

Myzhar, 2021-03-26:
The latest tests are ongoing... it will be released soon.
You can follow @Stereolabs3D on Twitter to be notified as soon as it will be available for download

closed 2022-04-27 by github-actions[bot]

Bovakinn, 2022-11-29:
@Myzhar, was that tutorial ever published? I can't seem to find it.

Myzhar, 2022-11-29:
@Bovakinn sure it is:
https://www.stereolabs.com/docs/sensors/magnetometer/

Bovakinn, 2022-11-29:
> @Bovakinn sure it is: https://www.stereolabs.com/docs/sensors/magnetometer/

Thanks, I assumed there was going to be a tutorial published on how to get a compass heading, not just reading the sensor values.

Myzhar, 2022-11-29:
Hi @Bovakinn 
the formulas to calculate the magnetic heading are complex and a tutorial explaining them is out of our scope.
You can find detailed papers with the full procedure online.
We prefer to apply those formulas inside the SDK and provide the calibrated heading data directly:
https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1MagnetometerData.html

Bovakinn, 2022-11-29:
@Myzhar, thank you so much. I don't know how, but I have never come across that page before. From navigating the site I've only ever seen these pages. 

https://www.stereolabs.com/docs/sensors/magnetometer/#output-data
https://www.stereolabs.com/docs/tutorials/using-sensors/

Myzhar, 2022-11-29:
That's part of the Python AP documentation. 
If you use Python language that's where you can find all the functions and variables explained in detail.