stereolabs/zed-python-api #177: cannot install python api

rsj007, 2021-04-06:
I am installing python api on my jetson tx2 by running "python3 get_python_api.py". However it seems that I can't connect to the download page. 

The output is below:
`vidia@nvidia:/usr/local/zed$ python3 get_python_api.py 
-> Downloading to '/usr/local/zed'
JETPACK 45
Detected platform: 
	 jetsons
	 Python 3.6
	 ZED SDK 3.4
-> Checking if https://download.stereolabs.com/zedsdk/3.4/jetsons/jp45/py36 exists and is available
Traceback (most recent call last):
  File "/usr/lib/python3.6/urllib/request.py", line 1325, in do_open
    encode_chunked=req.has_header('Transfer-encoding'))
  File "/usr/lib/python3.6/http/client.py", line 1281, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1327, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1276, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.6/http/client.py", line 1042, in _send_output
    self.send(msg)
  File "/usr/lib/python3.6/http/client.py", line 980, in send
    self.connect()
  File "/usr/lib/python3.6/http/client.py", line 1442, in connect
    server_hostname=server_hostname)
  File "/usr/lib/python3.6/ssl.py", line 407, in wrap_socket
    _context=self, _session=session)
  File "/usr/lib/python3.6/ssl.py", line 817, in __init__
    self.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 1077, in do_handshake
    self._sslobj.do_handshake()
  File "/usr/lib/python3.6/ssl.py", line 689, in do_handshake
    self._sslobj.do_handshake()
OSError: [Errno 0] Error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "get_python_api.py", line 270, in <module>
    urllib.request.urlretrieve(whl_file_URL, whl_file)
  File "/usr/lib/python3.6/urllib/request.py", line 248, in urlretrieve
    with contextlib.closing(urlopen(url, data)) as fp:
  File "/usr/lib/python3.6/urllib/request.py", line 223, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.6/urllib/request.py", line 532, in open
    response = meth(req, response)
  File "/usr/lib/python3.6/urllib/request.py", line 642, in http_response
    'http', request, response, code, msg, hdrs)
  File "/usr/lib/python3.6/urllib/request.py", line 564, in error
    result = self._call_chain(*args)
  File "/usr/lib/python3.6/urllib/request.py", line 504, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.6/urllib/request.py", line 756, in http_error_302
    return self.parent.open(new, timeout=req.timeout)
  File "/usr/lib/python3.6/urllib/request.py", line 526, in open
    response = self._open(req, data)
  File "/usr/lib/python3.6/urllib/request.py", line 544, in _open
    '_open', req)
  File "/usr/lib/python3.6/urllib/request.py", line 504, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.6/urllib/request.py", line 1368, in https_open
    context=self._context, check_hostname=self._check_hostname)
  File "/usr/lib/python3.6/urllib/request.py", line 1327, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [Errno 0] Error>`

What should I do to solve the problem?

adujardin, 2021-04-06:
You should check that the time and date are correctly set on your jetson, this can trigger certificate issues. You can also check if you can access the URL in a browser or using `wget`.

A workaround is to install the package manually, eg 

```

wget https://download.stereolabs.com/zedsdk/3.4/jetsons/jp45/py36 -O pyzed-3.4-cp36-cp36m-linux_aarch64.whl

python -m pip install ./pyzed-3.4-cp36-cp36m-linux_aarch64.whl

```

closed 2022-04-27 by github-actions[bot]

riccardokhm, 2023-02-17:
Same error here, on Windows 10! I'm getting this error:

urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:997)>

on line 175, urllib.request.urlretrieve(whl_file_URL, whl_file). 
Any suggestion? From my browser the access is denied.