stereolabs/zed-python-api #184: import error cannot import video camera from camera

omu34, 2021-05-19:
help with codes

qt-truong, 2021-05-19:
Hi @omu34,

Could you please specify your problem in more detail so that we can help you ? 
Thanks in advance.

omu34, 2021-05-19:
the problem is that the Html page shows only the icon of the images and
video but don't show the real image and video

qt-truong, 2021-05-19:
Thanks for getting back to us. I am sorry but I am not sure I understand your problem. 
Have you successfully managed to install the ZED SDK/Python API ? Are you trying to run a sample code ? 

I suggest you to visit our tutorial repository ([link](https://github.com/stereolabs/zed-examples/tree/master/tutorials)), this will guide you through your first steps with your camera.

closed 2022-04-27 by github-actions[bot]