stereolabs/zed-python-api #187: What is different between MEASURE.DEPTH and VIEW.DEPTH ?

KyuminHwang, 2021-07-18:
I want to get depth map through ZED camera

So, I am referring to the depth viewer and the sample code from https://github.com/stereolabs/zed-opencv/blob/master/python/zed-opencv.py.

However, the difference between the depth map (VIEW.DEPTH) and the saved depth map (MEASURE.DEPTH) seen in the depth viewer is too large. What is the problem?

[image] using **MEASURE.DEPTH**.

[image] using **VIEW.DEPTH**.

Myzhar, 2021-07-18:
`measure.depth` contains the depth values in floating point type. OpenCV converts them to a visualizable value before displaying.
`view.depth` does not contain depth values, but their normalized values in the range (0,255) to be rendered on screen.

KyuminHwang, 2021-07-18:
@Myzhar Thank you for timely response ! 👍 

You means that **view.depth** isn't contain any depth information? Then how display like depth map?
My guess is that I convert the RGB image to a GRAY Scale image, and arbitrarily normalize it to the [0,255] interval, is this a valid estimate?

The result of **VIEW.DEPTH** looks very similar to the depth map, so it feels like a visualization by estimating the depth information.

Could you please tell me which part of the code the library is?
(Then I will be able to clearly understand the rendering operation principle of **VIEW.DEPTH**.)

Myzhar, 2021-07-18:
`view.depth` is indeed the result of the normalization of `measure.depth`, mapping the values [0,max_range] -> [0,255].
The white pixels that you see while displaying `measure.depth` are relative to pixels with no depth information, so Nan or Inf.

KyuminHwang, 2021-07-19:
@Myzhar Thank you for comment 👍

closed 2021-07-19 by KyuminHwang