stereolabs/zed-python-api #190: Depth Estimation from Previously extracted data

faustoTapia, 2021-08-16:
Hi all,

I have collected stereo image pairs with ZED 1 (original and rectified) using  native OpenCV. I would like to use the python api to calculate the corresponding depth-maps offline (as I have access to a GPU only offline), is there an easy way to do so with the python API? 

I already got some depth-maps with OpenCV's stereo_SGBM. However, they are not as dense as the ones shown when I  use ZED_Depth viewer.

Thanks.

obraun-sl, 2021-08-16:
Hi,

Best way is to record a SVO file with ZED Explorer or the ZED SDK (see SVO Recording examples). 
Then you can process offline the SVO file as if it was a ZED camera connected.
See here : 
https://www.stereolabs.com/docs/video/recording/

faustoTapia, 2021-08-16:
Hi,

Wouldn't I need to have a GPU in the machine anyway to install ZED Explorer or the ZED SDK? In my case I do not have a GPU when I collect the data.

Thank you again.

obraun-sl, 2021-08-17:
Hi,
You can use ZED Explorer without a NVIDIA GPU. You just won't have access to all the compression mode (H264/H265 are based on NVIDIA encoding, but LOSSLESS is available even without NVIDIA GPU)

closed 2022-04-27 by github-actions[bot]