stereolabs/zed-python-api #193: No module named pyzed.sl on Jetson Nano

keli95566, 2021-09-03:
HI there! 
I am trying to get the Python streaming_sender.py to run on my newly flashed Jetson Nano, however, I am getting a ```Segmentation fault (core dumped)``` error. If I ran on gdb python debugger, I got the following messages: 
```Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
Traceback (most recent call last):
  File "streaming_sender.py", line 25, in <module>
    import pyzed.sl as sl
ImportError: No module named pyzed.sl
[Inferior 1 (process 28454) exited with code 01]```
telling me that they could not find the pyzed.sl file!
Other tool such as ZED_EXPLORE works fine.
```
System specifications:

1, Jetpack 4.5
2. Zed SDK 3.5
3. pyzed 3.5 ; downloaded with the get_pyzed.py script.

Anyone please help!! Thank you so much!!

qt-truong, 2021-09-06:
Hello @keli95566,

Could you please paste the log you get when running the script `get_python_api.py` ?
Also, if you used a virtual environment when installing the API, don't forget to activate it whenever you run a python script using pyzed.

closed 2022-04-27 by github-actions[bot]

fabrizioschiano, 2022-11-11:
@keli95566 did you find a solution to your issue?