stereolabs/zed-python-api #199: No code completion (Intellisense) for pyzed.sl

silegc-xguard, 2021-11-26:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi everyone,

I installed the Python API through the get_python_api.py script. While I can compile/run every example, I can't get code completion (Intellisense) to work. I'm running Python 3.8.5 with Anaconda in VsCode.

Is there a way to enable it ? It would be a very useful for development instead of always needing to refer to the online API documentation.

Thanks,
Simon

### Steps to Reproduce

1. Install the API using the get_python_api.py script
2. Try to use code completion in VsCode 

### Expected Result

Code completion should work

### Actual Result

No code completion results appear at any level of the API

### ZED Camera model

ZED2i

### Environment

```shell
OS: Windows 10
ZED SDK: v3.6.1
```

haavarpb, 2021-12-06:
Hi, I'll chip in with my solution here since I just had this problem myself. Please correct me if I am wrong, but I think this is due to the API being distributed as a .pyd-file. The VSCode intellisense is not able to read those directly, since its a binary file, more specifically a DLL. To get intellisense to work, I used the [genstub tool](https://mypy.readthedocs.io/en/stable/stubgen.html) to decompile the .pyd files present in the pyzed api and it created .pyi-files for my intellisense to read. Just run the genstub tool (I think I added `-m`-switch for module) and point it to the pyzed api folder. Voilá. Hope it helps!

silegc-xguard, 2021-12-06:
Thanks ! I got it to work that way.

closed 2022-04-27 by github-actions[bot]

junsukha, 2023-03-01:
@haavarpb @silegc-xguard  Hello, thanks for sharing the question and  tip. I'm following the guide but I don't understand the last part "point it to the pyzed api folder". Could you explain what that means?

kavehsgh, 2024-05-31:
Here is a very easy solution that I posted on the Stereolabs community website.
[https://community.stereolabs.com/t/no-auto-suggest-in-vscode-for-pyzed-sl/1633](url)