stereolabs/zed-python-api #200: Object ID keeps changing in non-changing environment

haavarpb, 2021-11-29:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi,

I am posting this here as I am using the python API, but I think this is issue is independent of the API as it is tied to using the core functionality. 

I am using ZED 2 for tracking human skeletons, and I notice that the object id's can change over time for the same object. Documentation is a bit limited, but in the object detection module on your site it is described that the object id will remain constant for a detected object. I have looked into using the uuid field instead, but those are constantly changing too. Is this intended behaviour? Is there something I can do to reduce the chance of the SDK recognizing a tracked object as a new object?

Regards, Håvard

### Steps to Reproduce

1. clone zed sdk samples
2. add a print statement to body_tracking.py to see object.id
3. run

### Expected Result

Object ID remains constant for a detected object

### Actual Result

Object ID sometimes changes

### ZED Camera model

ZED2

### Environment

```shell
OS: Windows 11, 64-bit 
Python version: 3.6.2
Pyzed: 3.6
```

### Anything else?

Perhaps expand documentation on topics around object detection: how it works, under which conditions and parameters that have impact on the quality of tracking results.

qt-truong, 2021-11-30:
Hi @haavarpb,

Thanks for reaching out to us. 

Indeed, a detected and tracked object should have a constant ID over time (you need to make sure the [`enable_tracking`](https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1ObjectDetectionParameters.html#a8dba66f45939e36a4e2e3eec371cad00) parameter is set to `True`). However, "ID switching" can indeed happen sometimes, and as you said it can be dependent on various conditions and parameters.
If it is possible for you, could you please send us an SVO where this phenomenon happens ? You can send it to support@stereolabs.com if you feel more comfortable

Thanks

haavarpb, 2021-12-15:
Hi, thanks for coming back to me.

I will come back to you with a recording if I come across something similar again, but for now, after tweaking the settings a bit and moving the camera to another position, it seems to have stabilized. Perhaps lighting also have an impact for tracking?

Regards

closed 2022-04-27 by github-actions[bot]