stereolabs/zed-python-api #202: Exception for broken SVO

jorgeuliana1, 2022-01-05:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

When opening an corrupted SVO with sl.Camera.open no exception is thrown. In Windows, the process keeps running (apparently it is IO bound forever). In Linux, the process is aborted. I tried a lot of thing, but all I can do is to have some workarounds done. It would be really nice if I could just have a try-except block to treat an eventual exception.

### Use-Case

It would make possible to just treat errors instead of just closing the script or having to close the terminal.

adujardin, 2022-01-05:
Thanks for the report, could you share an SVO file that trigger this error? You can send it to support@stereolabs.com if it's not meant to be public

jorgeuliana1, 2022-01-05:
Okay, I will send it to you right now. Thank you!

adujardin, 2022-01-18:
The ZED SDK 3.6.4 is released and includes the fix for this issue https://www.stereolabs.com/developers/release/

closed 2022-01-18 by adujardin