stereolabs/zed-python-api #204: Invalid URL for python API whl download

amaankhan02, 2022-01-23:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When following the instructions to download the ZED Python API, the link produced to download the .whl file is Invalid. 
Here is the link: https://download.stereolabs.com/zedsdk/3.6/ubuntu20/py38 - when going to the link directly, it throws a 404 Not Found error.

When checking this [github issue](https://github.com/stereolabs/zed-python-api/issues/179#issuecomment-836341554), @adujardin said to try this link https://download.stereolabs.com/zedsdk/3.5/ubuntu18/cu110/py37, however that link also fails and is invalid. 

Can someone please redirect us to a valid working link to download the latest python API for SDK 3.6.

### Steps to Reproduce

1. Run the get_python_api script which produces the following link: Go to https://download.stereolabs.com/zedsdk/3.6/ubuntu20/py38
2. Go to https://download.stereolabs.com/zedsdk/3.6/ubuntu20/py38
3.
...

### Expected Result

The whl file gets downloaded for pyzed

### Actual Result

404 Not Found

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 20 (also does not work on Windows 10)
ZED SDK VERSION: 3.6
Python Version: 3.8
```

adujardin, 2022-01-24:
There was an issue 2 days ago with our storage provider, that might explain this issue as the links are working correctly from my tests currently.
Just in case the direct link for :
- Ubuntu20 py3.8 ZED SDK 3.6 is https://stereolabs.sfo2.digitaloceanspaces.com/zedsdk/3.6/whl/ubuntu20/pyzed-3.6-cp38-cp38-linux_x86_64.whl
- Win10  py3.8 ZED SDK 3.6 https://stereolabs.sfo2.digitaloceanspaces.com/zedsdk/3.6/whl/win/pyzed-3.6-cp38-cp38-win_amd64.whl

For other versions, the URL follow the same pattern

amaankhan02, 2022-01-26:
Hi @adujardin,

I just tried pasting those 2 links in the browser right now and they still don't work and are leading to a 404 Not Found error. Is there a work around?

adujardin, 2022-01-26:
That's unexpected, we'll investigate how that's possible. Sorry about this

There's a workaround yes, you can compile the source of the wrapper, there are available here https://github.com/stereolabs/zed-python-api/tree/master/src

adujardin, 2022-01-26:
Your GitHub bio seems to indicate that you are located in the US, is this correct? That's would be even more surprising as the servers are in the bay area and accessible without issue even from Europe. Do you know if you have any proxy or firewall configuration that may interfere? 

Did you have issues with the C++ installers as well (they're hosted on the same server)?

amaankhan02, 2022-01-26:
@adujardin Yeah actually I think the problem was due to the proxy/firewall since I'm on my university wifi. Because my friend is now able to install it and he was not on the school wifi.

Sorry about that. So the link you gave is actually working well now. 

Thank you for your help!

closed 2022-01-26 by amaankhan02