stereolabs/zed-python-api #205: Unable to retrieve sensor data from .svo file

LoicFerrot, 2022-01-27:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I'm trying to access the IMU data when playing back a .svo file, but calling `imu_data.get_linear_acceleration()` always returns the same vector of `[4.585188705117234e-41, 0., 0.]` across different svo files. I also tried to get the temperature data, but it always returns -1. From what I understood of the documentation, only `get_linear_acceleration_uncalibrated` shouldn't be available in svo mode...

### Steps to Reproduce

```
from pyzed import sl
init_parameters = sl.InitParameters()
init_parameters.set_from_svo_file("/path/to/recording.svo")
zed = sl.Camera()
zed.open(init_parameters)
sensors_data = sl.SensorsData()
for i in range (100):
    if zed.grab() == sl.ERROR_CODE.SUCCESS:
        zed.get_sensors_data(sensors_data, sl.TIME_REFERENCE.IMAGE)
        imu_data = sensors_data.get_imu_data()
        temperature_data = sensors_data.get_temperature_data()
        print("imu data: ", imu_data.get_linear_acceleration())
        print("temperature data: ", temperature_data.get(sl.SENSOR_LOCATION.ONBOARD_LEFT))
```

### Expected Result

Real changing values, linear acceleration of norm 9.81 (camera is static)

### Actual Result

Constant values, not making sense

### ZED Camera model

ZED2i

### Environment

```shell
Latest docker `stereolabs/zed:3.6-gl-devel-cuda11.4-ubuntu20.04`
`torch==1.10.0+cu113`

NVIDIA GeForce RTX 3080
Intel® Core™ i9-10900K CPU @ 3.70GHz × 20
```

Myzhar, 2022-01-28:
Hi @LoicFerrot 
the SVO does not have acceleration and angular velocity data from the IMU, but only the fused attitude information.
That's because the IMU samples are provided at a higher framerate with respect to the camera frames (400 Hz vs max 100Hz) and saving them at a reduced sampled rate could generate unwanted effects like aliasing.

LoicFerrot, 2022-01-28:
Hey @Myzhar, thanks for your answer!
Do you have any idea on how I could retrieve the gravity vector of a static camera ?

I used the example code snippet from the documentation on  [Camera.get_position()](https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1Camera.html#af12b6dbe1e0b0349fe00a085c6a62eee) (which has to be slightly adapted), and as it seems expected since it gives the pose relative to the first frame, I only get a constant 0 translation and unit quaternion...

Myzhar, 2022-01-28:
That's not possible by using the SVO. 
You must write your own recorder for IMU data to save the information that you require at maximum rate retrieving them with the command `get_sensors_data` and the parameter `TIME_REFERENCE.CURRENT`.
Please note that the sensors' data grabbing must run in a separated thread with respect to the image data grabbing because the `grab` call is blocking.

closed 2022-01-28 by LoicFerrot