stereolabs/zed-python-api #206: ZED camera "CAMERA FAILED TO SETUP" error

kylecarow, 2022-02-02:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When running the `camera_control.py` example file with my ZED 1  (file [here](https://github.com/stereolabs/zed-examples/blob/master/camera%20control/python/camera_control.py)), I'm getting the following error:

```
Running...
Opening ZED Camera...
Grabber Error List
 -at: /builds/sl/ZEDKit/lib/src/sl_input/grabber/linux/linuxStereoGrabber.cpp_2625:  linux_grabber->construct return false
Grabber Error List
 -at: /builds/sl/ZEDKit/lib/src/sl_input/grabber/linux/linuxStereoGrabber.cpp_2625:  linux_grabber->construct return false
Grabber Error List
 -at: /builds/sl/ZEDKit/lib/src/sl_input/grabber/linux/linuxStereoGrabber.cpp_2625:  linux_grabber->construct return false
Grabber Error List
 -at: /builds/sl/ZEDKit/lib/src/sl_input/grabber/linux/linuxStereoGrabber.cpp_2625:  linux_grabber->construct return false
[ZED][Init][Info] Camera opening timeout reached
CAMERA FAILED TO SETUP
```

Any ideas?

### Steps to Reproduce

1. Verify camera is working in external camera app (ie Cheese)
2. Run `camera_control.py`

### Expected Result

Camera should open

### Actual Result

Camera fails to set up

### ZED Camera model

ZED

### Environment

```shell
OS: Ubuntu 18.04
```

obraun-sl, 2022-02-03:
This error means that it cannot create a video device for the camera. 
Can you post : 

--> Output of $ lsusb
 --> Output of ls /dev/video*

kylecarow, 2022-02-04:
`$ lsusb`
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 029: ID 2b03:f582  
Bus 002 Device 004: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 002 Device 011: ID 0461:4e2a Primax Electronics, Ltd 
Bus 002 Device 002: ID 1b1c:1b3d Corsair 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

`$ ls /dev/video*`
```
/dev/video0  /dev/video1
```

Yair-Ziv, 2022-02-06:
Have been getting the same error. 

### Description
When running our code we get the "CAMERA FAILED TO SETUP" error. We have been using the ZED Mini with the pyzed api.

### ZED Camera model
ZED Mini

### Environment 
Ubuntu server 20.04

### Additional information
1. Camera works fine when used with cheese or opencapture.
2. Camera does *not* work when used via the code **or** ZED\_Explorer.
3. lsusb shows the camera connected, as well as the HID interface.
4. ZED\_Diagnostics tool says it both didn't detect the camera even though ZEDDetected is true as well as detecting the device as ZED mini MCU; [diagnostics output](https://pastebin.com/WTJiMMDc).
5. Things we've tried and were unsuccessful - Reinstalling CUDA, replacing the camera with another ZED mini, rebooting the system, connecting directly to the camera (usually used with a hub).

Myzhar, 2022-02-06:
The latest Ubuntu update updated the v4l2 driver by introducing different management of some parameters that broke the communication with the ZED SDK.
We already found the cause of the problem and fixed it. A patch for the ZED SDK will be released in the next few days after completing the QA tests.

kylecarow, 2022-02-07:
Recently released version 3.6.5 fixes this issue!

closed 2022-02-07 by kylecarow

ymei, 2022-02-12:
> Recently released version 3.6.5 fixes this issue!

We just installed v3.6.5 on Ubuntu 18 (kernel version 5.4.0-99-generic).  The issue is still there (/builds/sl/ZEDKit/lib/src/sl_input/grabber/linux/linuxStereoGrabber.cpp_2625: linux_grabber->construct return false …). The camera be viewed by other generic UVC viewers. Any suggestions to solve this problem?

Myzhar, 2022-02-12:
Hi @ymei 
Please do not reply to closed issues. Always create a new one and tag the closed instead. 
Your problem seems something different, please create a new issue