stereolabs/zed-python-api #207: get point cloud data file (.pcd)

dohyeonYoon, 2022-02-03:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Hello
I am trying to get RGB image file (.jpeg) and point cloud data file (.pcd) through zed camera with python.

But in python api 
`point_cloud = sl.Mat()
zed.retrieve_measure(point_cloud, sl.MEASURE.XYZRGBA)

# Get the 3D point cloud values for pixel (i,j)
point3D = point_cloud.get_value(i,j)
x = point3D[0]
y = point3D[1]
z = point3D[2]
color = point3D[3]`

with the getValue() function as in the code
I could only get the pointcloud information of a specific pixel.

How can I save whole point cloud data in .pcd format with python?

I need the full pointcloud data file.

best regards,
dohyeon

obraun-sl, 2022-02-03:
Hi,

You can use the write() function of sl.Mat to write the point cloud in a file : 
https://www.stereolabs.com/docs/api/python/classpyzed_1_1sl_1_1Mat.html#ae0568cd94ee09fa51eaf50679e3a2def

--> point_cloud.write("test.pcd");

dohyeonYoon, 2022-02-03:
hello oliver

thank you so much !!!!

I solve the problem ! :)

best regards
dohyeon

closed 2022-02-08 by obraun-sl