stereolabs/zed-python-api #209: Rotating while positional tracking.

CreaM129129, 2022-03-17:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Why using any type of positional tracking samples (c++, python, both in spatial and positional tracking samples) camera thinks, it is rotating or moving. But in real life, camera is stationary. I tried recalibrating cammera and moving to differently lit environments, without any success. Sometimes it moves irradicly, sometimes it performs quite regural spiral/circle:
[image]
:

### Steps to Reproduce

1.Keeping camera in real life stationary.
2.Running any type of position-dependent demos.

### Expected Result

Position of camera in spatial_mapping/positional tracking is stationary.

### Actual Result

Camera in positional tracking moves irradicly/in a spiral

### ZED Camera model

ZED2

### Environment

```shell
OS: Ubuntu 20.04
CPU: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
GPU: NVIDIA GeForce MX450
ZED SDK VERSION: 3.7.0

Tried samples in both Python and c++ variant.
```

CreaM129129, 2022-03-21:
It seems position sensor is broken.

closed 2022-03-21 by CreaM129129

Myzhar, 2022-03-21:
@CreaM129129 have you written an email to support@stereolabs.com describing your problem?

CreaM129129, 2022-03-21:
Yes, this is why I closed the issue.
After _ZED_Calibration --calib_imu_  not fixing the problem, I assume position senzor on camera is broken.