stereolabs/zed-python-api #213: The camera frequency is set, but the time stamp when capturing the picture is incorrect

P970727, 2022-05-02:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Timestamp interval is not 1 / 30s，Inconsistent timestamp interval.

### Steps to Reproduce

from os import TMP_MAX
from time import monotonic
import pyzed.sl as sl
import cv2
import pandas as pd
import numpy as np

zed = sl.Camera()
# Set configuration parameters
init_params = sl.InitParameters()
init_params.camera_resolution = sl.RESOLUTION.HD1080 
init_params.camera_fps = 30 #相机的频率
#init_params.coordinate_units = sl.UNIT.METERS #坐标系单位
#init_params.depth_minimum_distance = 0.15 #Set the minimum depth perception distance to 15cm
#init_params.depth_maximum_distance = 20 #25m
init_params.depth_mode = sl.DEPTH_MODE.ULTRA
#init_params.sdk_gpu_id = 1#gpu

#Ultra模式提供最高的深度范围，并在感应范围内更好地保留Z精度

sp = sl.SensorParameters()
sp.sampling_rate = 0.9   #采样率
moi = sp.noise_density #白噪音
mwork = sp.random_walk  #随机游走
print(moi)
print(mwork)

#是否开启成功
err = zed.open(init_params)
if err != sl.ERROR_CODE.SUCCESS:
    print("没启动成功！")
    exit(-1)

# 传感器
sensors_data = sl.SensorsData()

#捕获图片
#image = sl.Mat()
left_image = sl.Mat()
depth_image = sl.Mat()

#imu表格格式
#右手系，后z，上y，左下。而EuRoc则是后-z，上x，左y。表格格式以EuRoc为主
name=['##timestamp [ms]','w_RS_S_x [rad s^-1]','w_RS_S_y [rad s^-1]','w_RS_S_z [rad s^-1]',
      'a_RS_S_x [m s^-2]','a_RS_S_y [m s^-2]','a_RS_S_z [m s^-2]']
csv=[]

key=''
i=0
print("按q----退出")
while i<200 :#按了q也不管用啊  key != 113

    if zed.grab() == sl.ERROR_CODE.SUCCESS:
    # A new image is available if grab() returns SUCCESS
        #获取图片（彩色图、深度图）
        zed.retrieve_image(left_image, sl.VIEW.LEFT) # Retrieve the left image
        zed.retrieve_image(depth_image, sl.VIEW.DEPTH) # Get a grayscale preview of the depth map

        #提取imu数据
        zed.get_sensors_data(sensors_data, sl.TIME_REFERENCE.IMAGE) # Retrieve only frame synchronized data
        imu_data = sensors_data.get_imu_data()
        
        #获得时间戳
        timestamp = zed.get_timestamp(sl.TIME_REFERENCE.IMAGE) 
        #print("Image resolution: ", image.get_width(), " x ", image.get_height()," || Image timestamp: ", timestamp.get_milliseconds())
        
        #存图片
        #j = str(i) +'zed'

        img_left = left_image.get_data() 
        img_depth = depth_image.get_data() 

        # img = np.hstack([img_left, img_depth])
        # img = cv2.resize(img, (800, 400))
        # cv2.imshow("ZED-depth", img)
        #cv2.imwrite("./picture/"+j+'.png',img)
        cv2.imwrite('./picture/image/{:06}.png'.format(i),img_left)
        cv2.imwrite('./picture/depth/{:06}.png'.format(i),img_depth)
        

        #获得imu的角速度、加速度，并存入csv列表中
        # Retrieve linear acceleration and angular velocity
        angular_velocity = imu_data.get_angular_velocity()#3x1 角速度
        # tmp = angular_velocity[0]
        # angular_velocity[0] = angular_velocity[1]
        # angular_velocity[1] = tmp
        # angular_velocity[2] = 0-angular_velocity[2]
        
        linear_acceleration = imu_data.get_linear_acceleration()#3x1  线加速度
        # tmp = linear_acceleration[0]
        # linear_acceleration[0] = linear_acceleration[1]
        # linear_acceleration[1] = tmp
        # linear_acceleration[2] = 0-linear_acceleration[2]

        res = []
        res.append(timestamp.get_milliseconds())#毫秒
        res.extend(angular_velocity)
        res.extend(linear_acceleration)
        csv.append(res)
       
       
        key = cv2.waitKey(5)
        print(key)
        
    #更新帧数
    i=i+1

df = pd.DataFrame(columns=name, data=csv)
df.to_csv('./picture/imu.csv', index=False)#保存imu表格数据
cv2.destroyAllWindows()
zed.close()

### Expected Result

The interval between timestamps is fixed at 1 / 30s

### Actual Result

Timestamp interval is not 1 / 30s，Inconsistent timestamp interval.

### ZED Camera model

ZED2

### Environment

```shell
ubuntu18.04
zed sdk 3.7.2
```

fennecinspace, 2022-06-07:
I'm trying to sync multiple SVO files a remove excess frames from start and end, to have same length synched recordings from multiple ZED cameras. and this issue is stopping me from doing that.

fennecinspace, 2022-07-15:
Exists still.

closed 2022-08-21 by github-actions[bot]