stereolabs/zed-python-api #214: pyzed.slSensorsData has no attirbute MagnetometerData

Bovakinn, 2022-05-17:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi I recently upgraded the SDK to version 3.7.2 for Jetpack 4.4 on an Nvidia Jetson Xavier NX and encounter the following error when trying to access the Magnetometer using the example python script.  pyzed.slSensorsData has no attirbute MagnetometerData

### Steps to Reproduce

Installed this package: https://download.stereolabs.com/zedsdk/3.7/l4t32.4/jetsons
Used code in tutorial 7: Getting Sensors Data

### Expected Result

Returned Magnetometer readings 

### Actual Result

 pyzed.slSensorsData has no attirbute MagnetometerData

### ZED Camera model

ZED2

### Environment

```shell
Nvidia Xavier NX
arm64
Jetpack version is 4.4-b144
Python version is 3.6.9
Cuda version is 10.2.89
```

alassagne-sl, 2022-05-17:
Hello, are you talking about [this tutorial ?](https://github.com/stereolabs/zed-examples/blob/master/tutorials/tutorial%207%20-%20sensor%20data/python/sensor_data.py)
On my setup it runs correctly.
Maybe you just have a typo somewhere ?
you should write `sensors_data.get_barometer_data()` and not `sensors_data.MagnetometerData`.

If that's not the issue, can you share your code ?

Bovakinn, 2022-05-18:
I'm not sure where I got the method from, but the tutorial you linked worked perfectly.

closed 2022-05-18 by Bovakinn

Bovakinn, 2022-05-18:
The code I was using was found on this page https://www.stereolabs.com/docs/sensors/magnetometer/

alassagne-sl, 2022-05-18:
Thank you for the report ! We'll change this as soon as we can.
Have a great day
Antoine