stereolabs/zed-python-api #215: Python API for 3.7.4 on Jetpack 5.0

bblumberg, 2022-06-03:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am unable to install the Python API for 3.7.4 on a AGX Orin running Jetpack 5.0. When I run get_python_api.py, I get: Unsupported l4t/jetpack version

Is there something I can do on my end to work around this problem?

Thanks,

bb

### Steps to Reproduce

1. cd /usr/local/zed
2 python3 get_python_api.py
3.
...

### Expected Result

Python wrapper gets installed.

### Actual Result

Unsupported l4t/jetpack version

### ZED Camera model

ZED2

### Environment

```shell
ZED 3.7.4
AGX Orin
JetPack 5.0.1
```

### Anything else?

I'd love to be able to use the camera on the Orin.

alassagne-sl, 2022-06-03:
Hello and thank you for reaching us out.
There maybe an issue on our side, since we do support your configuration. Can you send us the full log ?
In the meantime, if you are in a hurry, I can give you a direct link to the whl file. The full log will give me the python version that you need.

Antoine

bblumberg, 2022-06-03:
> Thanks for the quick response. It would be great if you could send me a
> direct link:
>
> Here is the log from the terminal (is there a more complete log somewhere?)
>
> python3 get_python_api.py
> -> Downloading to '/usr/local/zed'
> Unsupported l4t/jetpack version
>
> Here is my python
> Python 3.8.10 (default, Mar 15 2022, 12:22:08)
> [GCC 9.4.0] on linux
>
> Thanks again,
>
> bb

Ok, it is a little logical bug in the get_python_api.py script: Lines
231-233 should be removed or commented out, e.g.:

#else:
            #    print('Unsupported l4t/jetpack version')
            #    sys.exit(1)

bobedwards123, 2022-06-04:
Can you please post the whl file I have the same issue, am unable to install the python API on this version of jetpack. Thanks

bblumberg, 2022-06-04:
I am away from my computer but it’s an easy fix. Go to the getpythonapi.py
file in /usr/local/zed and comment out the lines in question.

bb

alassagne-sl, 2022-06-07:
Indeed, it's even easier to comment the lines 231 to 233 and run the script again. It will be fixed in the next version.

closed 2022-06-07 by alassagne-sl

ardasevinc, 2022-06-14:
This issue should not be closed as it's not fixed. I just faced this issue in my Jetpack 5.0.1, ZED 3.7.4 install. If the issue remains open, others who face this issue can solve it faster before the next version comes out. Thanks.

bblumberg, 2022-10-11:
Thanks for the quick response. It would be great if you could send me a
direct link:

Here is the log from the terminal (is there a more complete log somewhere?)

python3 get_python_api.py
-> Downloading to '/usr/local/zed'
Unsupported l4t/jetpack version

Here is my python
Python 3.8.10 (default, Mar 15 2022, 12:22:08)
[GCC 9.4.0] on linux

Thanks again,

bb