stereolabs/zed-python-api #219: import error when run as superuser for python 3.8 on ubuntu

chensley42, 2022-09-26:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I've just made a clean install of the latest version of the ZED api on my x86 system running Ubuntu 20.04 with Cuda 11.7. I downloaded the runfile for the appropriate release of the latest api version and installed it without issue, including the python API. Normally, I have no problem importing and using the python api. However, I have some scripts using the camera which require to be run with sudo in order to access some other modules. With the prior version of the python api (for cuda 11.4) this was no problem, but the new version throws an importation error for an undefined symbol when running the python script as root. The issue is as follows:

>>>import pyzed.sl as sl
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError:  /usr/local/lib/python3.8/dist-packages/pyzed/sl.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN2sl14InitParametersC1ENS_10RESOLUTIONEibNS_10DEPTH_MODE_4UNITENS_17COORDINATE_SYSTEMbiffbibNS_6StringEiP8CUctx_stNS_9InputTypeES5_bb

Since this functionality had worked fine in the past, and it seemed like an issue with the cpython links directly, I figured I should file a report and also ask if there was a workaround for now.

### Steps to Reproduce

1. Install ZED API Version 3.7.7 for Ubuntu 20
2. run python 3.8.10 shell with sudo ("sudo python" or "sudo python3")
3. type "import pyzed.sl"

### Expected Result

module imports successfully

### Actual Result

ImportError:  /usr/local/lib/python3.8/dist-packages/pyzed/sl.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN2sl14InitParametersC1ENS_10RESOLUTIONEibNS_10DEPTH_MODE_4UNITENS_17COORDINATE_SYSTEMbiffbibNS_6StringEiP8CUctx_stNS_9InputTypeES5_bb

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu Bionic
CPU: X86_64
GPU: Nvidia 1060 Mobile
ZED SDK Version: 3.7.7
Python Version: 3.8.10
Cuda Version 11.7
```

closed 2022-11-01 by github-actions[bot]

fabrizioschiano, 2022-11-11:
@chensley42 , did you find a solution to this issue?

ruoshiwen, 2024-08-12:
I am running sdk in a container. 
But I have the same issue. Any idea please?