stereolabs/zed-python-api #222: Unable to install Zed Python API

gautamigolani, 2023-03-19:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi! I'm trying to install the Zed Python API as per the instructions on the github page to use the Zed 2 but I am facing an issue while doing so.

### Steps to Reproduce

1. Download Zed SDK, CUDA and other dependencies
2. Run the install script
3.
...

### Expected Result

C:\Program Files (x86)\ZED SDK>python get_python_api.py

-> Downloading to 'C:\Program Files (x86)\ZED SDK'
Detected platform:
         win_amd64
         Python 3.9
         ZED SDK 3.8
-> Checking if https://download.stereolabs.com/zedsdk/3.8/whl/win_amd64/pyzed-3.8-cp39-cp39-win_amd64.whl exists and is available
-> Found ! Downloading python package into C:\Program Files (x86)\ZED SDK\pyzed-3.8-cp39-cp39-win_amd64.whl
-> Installing necessary dependencies
Requirement already satisfied: numpy in c:\users\gauta\appdata\local\programs\python\python39\lib\site-packages (1.24.2)
Processing c:\program files (x86)\zed sdk\pyzed-3.8-cp39-cp39-win_amd64.whl
Collecting numpy>=1.13
  Using cached numpy-1.24.2-cp39-cp39-win_amd64.whl (14.9 MB)
Collecting cython>=0.28
  Using cached Cython-0.29.33-py2.py3-none-any.whl (987 kB)
Installing collected packages: numpy, cython, pyzed
Successfully installed cython-0.29.33 numpy-1.24.2 pyzed-3.8
Done
Installing OpenGL dependencies required to run the samples
-> Downloading PyOpenGL-3.1.5-cp39-cp39-win_amd64.whl
Processing c:\program files (x86)\zed sdk\pyopengl-3.1.5-cp39-cp39-win_amd64.whl
PyOpenGL is already installed with the same version as the provided wheel. Use --force-reinstall to force an installation of the wheel.
-> Downloading PyOpenGL_accelerate-3.1.5-cp39-cp39-win_amd64.whl
Processing c:\program files (x86)\zed sdk\pyopengl_accelerate-3.1.5-cp39-cp39-win_amd64.whl
PyOpenGL-accelerate is already installed with the same version as the provided wheel. Use --force-reinstall to force an installation of the wheel.
Pyzed directory is c:\program files\python39\lib\site-packages
  To install it later or on a different environment run :
 python -m pip install --ignore-installed C:\Program Files (x86)\ZED SDK\pyzed-3.8-cp39-cp39-win_amd64.whl

### Actual Result

C:\Program Files (x86)\ZED SDK>python get_python_api.py

-> Downloading to 'C:\Program Files (x86)\ZED SDK'
Detected platform:
         win_amd64
         Python 3.9
         ZED SDK 3.8
-> Checking if https://download.stereolabs.com/zedsdk/3.8/whl/win_amd64/pyzed-3.8-cp39-cp39-win_amd64.whl exists and is available
-> Found ! Downloading python package into C:\Program Files (x86)\ZED SDK\pyzed-3.8-cp39-cp39-win_amd64.whl
-> Installing necessary dependencies
Requirement already satisfied: numpy in c:\users\gauta\appdata\local\programs\python\python39\lib\site-packages (1.24.2)
Processing c:\program files (x86)\zed sdk\pyzed-3.8-cp39-cp39-win_amd64.whl
Collecting numpy>=1.13
  Using cached numpy-1.24.2-cp39-cp39-win_amd64.whl (14.9 MB)
Collecting cython>=0.28
  Using cached Cython-0.29.33-py2.py3-none-any.whl (987 kB)
Installing collected packages: numpy, cython, pyzed
Successfully installed cython-0.29.33 numpy-1.24.2 pyzed-3.8
Done
Installing OpenGL dependencies required to run the samples
-> Downloading PyOpenGL-3.1.5-cp39-cp39-win_amd64.whl
Processing c:\program files (x86)\zed sdk\pyopengl-3.1.5-cp39-cp39-win_amd64.whl
PyOpenGL is already installed with the same version as the provided wheel. Use --force-reinstall to force an installation of the wheel.
-> Downloading PyOpenGL_accelerate-3.1.5-cp39-cp39-win_amd64.whl
Traceback (most recent call last):
  File "C:\Program Files (x86)\ZED SDK\get_python_api.py", line 207, in <module>
    install_win_dep("PyOpenGL_accelerate", py_vers)
  File "C:\Program Files (x86)\ZED SDK\get_python_api.py", line 61, in install_win_dep
    urllib.request.urlretrieve(whl_file_URL, whl_file)
  File "C:\Users\gauta\AppData\Local\Programs\Python\Python39\lib\urllib\request.py", line 278, in urlretrieve
    raise ContentTooShortError(
urllib.error.ContentTooShortError: <urlopen error retrieval incomplete: got only 65065 out of 366922 bytes>

### ZED Camera model

ZED2

### Environment

```shell
OS: Windows 10
CUDA 11.7
Python 3.9.5
ZED SDK 3.8.2
```

woodrow-lwd, 2023-07-12:
the same.why nobody answer? i can't understand. if they could just list whl files on webside, and with details about suitable python version,cuda version etc. It should be easier for us user.

zhangYQHBAU, 2023-10-28:
When installing on an ubuntu system, the command used when compiling is： python 3 setup. py build, display error: pyzed/sl. pyx: 10455:161 "c_position_type" is not a type identifier, how to solve this problem?