stereolabs/zed-python-api #226: ImportError: DLL load failed while importing sl: The specified procedure could not be found.

felixcrisan420, 2023-06-29:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

After a number of attempts to run the hello_zed.py file, I am met with this message:
ImportError: DLL load failed while importing sl: The specified procedure could not be found.

How can I fix this?

### Steps to Reproduce

The error persitst at other versions aswell

### Expected Result

The hello_zed.py to work.

### Actual Result

It spit out the error:  ImportError: DLL load failed while importing sl: The specified procedure could not be found.

### ZED Camera model

ZED2i

### Environment

```shell
OS: Windows 11
Graphics: Nvidia RTX 2070 with Max-Q Design (Mobile)
CUDA: 12.0
Python: 3.10
Zed SDK: 4.0
```

jquintanilla4, 2023-07-04:
I have this same issue but on the following:
OS: Windows 10
Graphics: Nvidia 3090
CUDA: 11.7
Python: 3.8.16
ZED SDK: 4.0

chopsuey5000, 2023-09-26:
Having same issue here. Any news on this?

chopsuey5000, 2023-09-26:
I just install SDK v4.0.7 with cuda 1..8 and works like a charm.