stereolabs/zed-python-api #248: Request for .pyi stub files

haavarpb, 2025-04-16:
### Preliminary Checks

- [x] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [x] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Hi,

It would be really nice if you could maintain python stub files (.pyi) for pyzed.

After looking around I have found a tool `stubgen` provided by `mypy` that can generate stubfile _drafts_. It works to a degree, but it is not perfect. Working in VSCode, I have checked the Pylance documentation, and while they have the "python.analysis.useLibraryCodeForTypes" option, it is not really full fledged. On their github, it seems they won't support this any time soon https://github.com/microsoft/pylance-release/issues/674. They point to library maintainers, or even the Cython community. In fact, there is work done on the topic of generating stubfiles based on cython files https://github.com/cython/cython/pull/5744.

Therefore, I turn to you directly. I would like to make the argument that as long as you put an effort to supporting Python it would make sense to maintain a stub file too. I suppose the Python support is made because you'd like to cater the masses, so why not make their lives a little easier. I myself is currently doing a PhD. in robotics and I am prototyping with the ZED Mini. However, I find myself getting lost in the online documentation trying to find which object had which property. I spend quite a lot of time on there, and in my subjective opinion it causes some friction.

Thanks for hearing me out. Besides that, you have made a great product!

Cheers

haavarpb, 2025-04-16:
Related #199

adujardin, 2025-05-07:
Hi, that's a good suggestion, we'll include a stub in the package from now on. Thanks for the report

seralexeev, 2025-08-05:
Hey @adujardin - thanks for including the stub in the package, it's super helpful!

Would it also be possible to include comments as docstrings for classes and members?

That would really improve the dev experience with autocompletion and in editor docs.